bblanchon/ArduinoJson #1817: Json very nesting - Filter

IsaiasGatica, 2022-10-27:
Hello first great library!
I was trying to deserialize a very nested json and due to the large size I tried to use the filters. But it generated me an even bigger Jsondynamic.

- For example the filter:

`JsonObject filter_Children_0_Children_0_Children_0_Children_0_Children_0 = filter["Children"][0]["Children"][0]["Children"][0]["Children"][0]["Children"].createNestedObject();
filter_Children_0_Children_0_Children_0_Children_0_Children_0["id"] = true;
filter_Children_0_Children_0_Children_0_Children_0_Children_0["Value"] = true;`

Am I applying it wrong?

On the other hand, is it possible to filter, for example, an id of a "branch" and not all the ids of all the children?

Sorry if it is not understood, my english is not very good. 
Thanks, greetings!

### Json nesting= [Json.zip](https://github.com/bblanchon/ArduinoJson/files/9882514/Json.zip)

bblanchon, 2022-10-29:
Hi @IsaiasGatica,

I'm sorry, I don't understand what you mean by "branch".
It's impossible to do a recursive filter if that's what you meant.
Also, the filter cannot apply different rules to each element of an array (#1316).

Best regards,
Benoit

IsaiasGatica, 2022-10-30:
Hello, thank you for replying! I attach an image to highlight what I call "branch" 😅

[image]

I was analyzing the examples. From the web and from the book, and I think the problem is how is my Json (attached). 
All the children have the same name.
The problem is similar to (#1316).

Anyway I was able to reduce the size.
Thank you very much!

bblanchon, 2022-10-31:
You're welcome. I'm glad I could help :)
Shall we close this issue then?

closed 2022-10-31 by IsaiasGatica