bblanchon/ArduinoJson #1818: Iterating over parsed json

HamzaHajeir, 2022-10-30:
Hi there,

I'm looking for a way to parse JSON into holders, It's providable by a class to be responsible for different forms of formats.

The class returns the parsed items within a vector: `vector<pair<string,string>>`.

How can I populate the vector without needing to know keys.

I'm looking for something like:
```cpp
vector<pair<string,string>> parsed_list;
for (auto j : parsedJson){
    parsed_list.push_back(make_pair(j.key, j.value));
}
```

Inversely - here's an example code to serialize provided `vector<pair<string,string>>`:
```cpp
std::string JSON::Serialize(const VPSS &data){
    DynamicJsonDocument doc(2048);
    for (auto d : data){
        doc[d.first] = d.second;
    }
    std::string rv = doc.as<std::string>();
    return rv;
}
```

Thank you

HamzaHajeir, 2022-10-30:
I think with a bit better search, I've found it here:

https://arduinojson.org/v6/api/jsonobject/begin_end/

My code goes:
```cpp
VPSS JSON::Deserialize(const std::string &data){

    DynamicJsonDocument doc(1024);
    deserializeJson(doc, data);
    JsonObject root = doc.as<JsonObject>();

    VPSS rv;
    
    for (JsonPair kv : root) {
        rv.push_back(std::make_pair(kv.key().as<std::string>(), kv.value().as<std::string>()));
    }
    return std::make_pair((rv.size() ? Format_onSuccessfulDeserialize : Format_onFailedDeserialize), rv);
}
```

closed 2022-10-30 by HamzaHajeir

HamzaHajeir, 2022-10-30:
**UPDATE**

Corrected Deserialization Function, due to compilation errors:
```cpp
Deserialize_Return JSON::Deserialize(const std::string &data){

    DynamicJsonDocument doc(1024);
    deserializeJson(doc, data);
    JsonObject root = doc.as<JsonObject>();
    VPSS rv;

    for (JsonPair kv : root){
        rv.push_back(std::make_pair(kv.key().c_str(), kv.value().as<std::string>()));
    }
    return std::make_pair((rv.size() ? Format_onSuccessfulDeserialize : Format_onFailedDeserialize), rv);
}
```

bblanchon, 2022-10-31:
Hi @HamzaHajeir,

> I'm happy to receive any comments there.

This looks correct, but I'm surprised you chose a `vector<pair<string,string>>` over a `map<string,string>`.
Also, I'd use `emplace_back()` instead of `push_back(make_pair())`.

Best regards,
Benoit

HamzaHajeir, 2022-10-31:
You're correct, I don't see a case where duplicated keys is present, so a
map is the way to go.

To my knowledge, emplace_back would require a vector with enough empty
size.. isn't it?

With thanks,
Hamzah