bblanchon/ArduinoJson #1819: Deserialize from json object to C++ class object

mistergreen, 2022-11-01:
Hi,
Just curious if there's a function in ArduinoJson to map a json object into a C++ class object, like modifying the values of an C++ object.
Say I have a json
```
{
   "Object1":{
      "id":123,
      "name":"Sensor"
   }
}
```
I want the key and value to automatically transfer into an object of class
```
class Object1 {
  public:
     int id;
     string name;
}
```

I know I can manually deserialize the json and set the values like

```
DynamicJsonDocument doc(1024);
deserializeJson(doc, json);
Object1 object1;
object1.id= doc["id"];
object1.sensor = doc["sensor"];
```

I was wondering if there's a function to map this or if I need to look for or create a mapping helper.
Thanks.

bblanchon, 2022-11-02:
Hi @mistergreen,

You'll find the answer here: 

* [Can I automatically serialize an object?](https://arduinojson.org/v6/faq/automatically-serialize-an-object/)

Best regards,
Benoit

mistergreen, 2022-11-02:
Thanks. I'll have to figure out a homegrown way to do it like creating a json header file to define members and data type instead of blindly stepping members.

closed 2022-11-02 by mistergreen