bblanchon/ArduinoJson #1821: Can a JSON document compressed directly?

ElcRcp, 2022-11-03:
Hello, I wonder if it is poisible to compress a JSON document? 
I gather large amount of data and planning to use miniz library before sending data over internet and I'm avare that I can serialize document into a string or file and compress it but it is not looking efficient so what I'm seeking is compressing document directly. Is it possible?

bblanchon, 2022-11-04:
Hi @ElcRcp,

You can write [a custom writer](https://arduinojson.org/news/2019/11/01/version-6-13-0/#custom-writer) to interface ArduinoJson with miniz.

Best regards,
Benoit

ElcRcp, 2022-11-04:
Hi @bblanchon 
Thanks for the fast reply. As I understand, using a custom writer I'll be able to directly serialize a document to compression function input and I'll get a data array as a result but since result is not a JSON Document anymore, it cannot be used with streamutils features right?

As my understanding; using a custom writer with compressing is almost same with seralizing document into an array and then compressing it, am I correct?

One last thing, what would be if I try to add an compressed data into a json document? Is there a value size limit ? For example if I have a "uint8_t * compItem" with size of 100kBytes would "doc["key"] = compItem;" work? This way I can transfer data as a json document, receipent would easly parse the document and decompress the key's value.

bblanchon, 2022-11-08:
StreamUtils works with the `Stream` interface whether it transports JSON or not.
You can, therefore, use this library after compressing the data.

Using a custom writer will use less RAM since you don't need to store the uncompressed document.

You can pass binary data in a JSON string with `doc["key"] = JsonString(compItem, size)` but make sure the receiver doesn't try to decode UTF-8 characters.

ElcRcp, 2022-11-09:
Thank you @bblanchon, you were a great help again !

closed 2022-11-09 by ElcRcp