bblanchon/ArduinoJson #1823: How to parse a large JSON PROGMEM char array in chunks?

tobiasfaust, 2022-11-11:
Hi, i have a large fix json document as progmem char array (short example)
```
const char JSON[] PROGMEM = R "=====( {
	"Solax-X1": {
		"data": {
			"livedata": [{
					"position": [0, 1],
					"name": "GridVoltage",
					"realname": "Grid Voltage",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{
					"position": [2, 3],
					"name": "GridCurrent",
					"realname": "Grid Current",
					"datatype": "float",
					"factor": 0.1,
					"unit": "V"
				},
				{....}
			]
		}
	}
}
```

I want to process them in chunks, but your example dont work because what is the inputstream?
Its possible to process PROGMEM Strings in chunks?

```
StaticJsonDocument<32> filter;
filter["name"] = true;

input.find("\"characters\":[");
do {
    deserializeJson(doc, input, DeserializationOption::Filter(filter));
    // ...
} while (input.findUntil(",","]"));

```

Currently i get an Memory error:

```
StaticJsonDocument<2048> regjson;
StaticJsonDocument<200> filter;

filter[this->InverterType]["data"]["livedata"] = true;

DeserializationError error = deserializeJson(regjson, JSON, DeserializationOption::Filter(filter));

 if (!error) {
      // Print the result
     [....]
   } else {
      Serial.print("Failed to parse JSON Register Data: "); Serial.print(error.c_str());
   }

```

bblanchon, 2022-11-14:
Hi @tobiasfaust,

The [deserialization in chunks](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/#deserialization-in-chunks) technique relies on the `Stream` interface.
In other words, you need a `Stream` instance to use this technique.

I added [`ProgmemStream`](https://github.com/bblanchon/ArduinoStreamUtils#how-to-use-progmem-as-a-stream) in StreamUtils 1.7.0 for this purpose.
I just pushed it, so it might not be immediately available in the Library Manager.

I'll let you figure out the details, but basically, you must create a `ProgmemStream` and pass it to `deserializeJson()`:

```c++
ProgmemStream input(JSON);
DeserializationError error = deserializeJson(regjson, input, DeserializationOption::Filter(filter));
```

Best regards,
Benoit

tobiasfaust, 2022-11-18:
Hi Benoit,
amazing! You are the greatest :) It works perfectly. So i could reduce my memory from 5k to 0.5k by enabling the option to enhance my json array with more objects in future :)
If you or others like, you are able to find the new PROGMEM Stream code in my modbus.cpp below row 300. 

https://github.com/tobiasfaust/SolaxModbusGateway

closed 2022-11-18 by tobiasfaust