bblanchon/ArduinoJson #1825: Json file doesn't be saved with the new value

pepson84, 2022-11-17:
**Description**
hello there. 
what i would like to do is to update in the main loop the values of some variables contained in a json file. but it doesn't work

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns deserialization
3. <code>deserializeJson()</code> returns <code>IncompleteInput</code>
4. Input comes from a pointer
5. Increasing the buffer size doesn't solve the issue
6. Reading from stream doesn't fix the issue

**Environment**
* Microcontroller: ESP8266
* Core/Framework: ESP8266 core 3.0.2 using Arduino  IDE 1.8.19
* IDE: Arduino IDE 1.8.19

**Reproduction code**
```c++
#define BLYNK_TEMPLATE_ID "XXXXXX"
#define BLYNK_DEVICE_NAME "Meter"
#define BLYNK_FIRMWARE_VERSION        "10.0.0"

#define BLYNK_PRINT Serial

//#define BLYNK_DEBUG

#include <Nextion.h>

#define APP_DEBUG

// Uncomment your board, or configure a custom board in Settings.h
//#define USE_SPARKFUN_BLYNK_BOARD
#define USE_NODE_MCU_BOARD
//#define USE_WITTY_CLOUD_BOARD
//#define USE_WEMOS_D1_MINI

#include "BlynkEdgent.h"

BlynkTimer timer;
float h=0;
float t=0;
float old_h=0;
float old_t=0;
float  enerday;
float enertot;
const char *t0h = "T_mis.txt=\"";
const char *h0h = "HR.txt=\"";

#include <SoftwareSerial.h>

int conta;

 #include <ArduinoJson.h>

 #include <StreamUtils.h>
 // const size_t capacity = JSON_OBJECT_SIZE(1500)+16;
   //ok
 DynamicJsonDocument doc(1400);//modifica del 15/11/22 (spostato dal loop qui)

//EepromStream eepromStream(300, 4000); //with this wifi settings has been kept but it doesn't upgrade the //json
//EepromStream eepromStream(0, 200);//it lost wifi settings ,but it upgrade the json

// This function sends Arduino's uptime every second to Virtual Pin 2.
void myTimerEvent()
{
  // You can send any value at any time.
  // Please don't send more that 10 values per second.
  Blynk.virtualWrite(V2, millis() / 1000);
  textLoop();
}
void setup()
{
  Serial.begin(115200);
  delay(100);
mySerial.begin(115200);
BlynkEdgent.begin();
   timer.setInterval(1000L, myTimerEvent);

 deserializeJson(doc, eepromStream); 
   Serial.println("doc nel loop");
serializeJsonPretty(doc, Serial);
 

//https://arduinojson.org/v6/api/jsondocument/memoryusage/
//Serial.println("memoryUsage(): ");
//Serial.println(doc.memoryUsage());

   EEPROM.begin(4096);
     Serial.println(" eeprom ispnibile setup");
 Serial.println( EEPROM.length());
}

void loop() {
  
BlynkEdgent.run();

  timer.run();
 
  if(conta=1)
{ 

enerday=4.5;

}

conta++;
enerday+=enerday;
enertot+=enerday*2;

if(conta%2==0){
  Serial.println("conta: "); 
Serial.println(conta);

 //deserializeJson(doc, eepromStream);
 doc["enerday"]=enerday;
 doc["enertot"]=enertot;
 doc["gennaio"]=enertot*2;
 doc["autoc_gennaio"]=enertot*3;
 doc["enerim_gennaio"]=enertot*4;
 delay(100); 
serializeJson(doc, eepromStream);
serializeJsonPretty(doc, Serial);
EEPROM.commit();
delay(2000); 
  Serial.println(" eeprom dispnibile LOOP");
 Serial.println( EEPROM.length());
}
delay(2000); 
Serial.println("memoryUsage(): ");
Serial.println(doc.memoryUsage());
}
```

**Remarks**
it seems that in some address of the heap it can save right the wifi settings, but not the json "doc" i want to use to store and upgrade the variables. 

thanks for your support.

bblanchon, 2022-11-18:
Hi @pepson84,

There are too many moving parts in this program.
Please remove every irrelevant line and keep it under 20 lines.

In your message, please also include a comparison of what works and what doesn't.

Best regards,
Benoit

pepson84, 2022-11-18:
> Hi @pepson84,
> 
> There are too many moving parts in this program. Please remove every irrelevant line and keep it under 20 lines.
> 
> In your message, please also include a comparison of what works and what doesn't.
> 
> Best regards, Benoit

Hello @bblanchon,
thanks, i removed the irrelevant line.

bblanchon, 2022-11-20:
I'm puzzled because the Troubleshooter report says, "deserializeJson() returns IncompleteInput", but the program doesn't contain any call to `deserializeJson()`.

Anyway, the only thing I noticed on your program is the missing definition of `eepromStream`.
You must declare `eepromStream` inside the `loop()` function, preferably on the line above `serializeJson()`; otherwise, it will append instead of overriding.
In other words, the `EepromStream` doesn't automatically rewind, so you need to create a new one.
[*Be kind rewind*](https://youtu.be/_NRoUAe3NW0) 😉

bblanchon, 2022-11-25:
I don't know why the Blynk configuration is lost; that's probably a question you should address with Blynk's support team.

BTW, there is no hardware EEPROM on the ESP8266.
Instead, the EEPROM library emulates the behavior using a flash sector (hence the additional `commit()` function).
I suggest you try using LittleFS or SPIFFS to store your JSON document.
Hopefully, this will solve all your issues.

pepson84, 2022-11-27:
i'll let you know; thank you so much for your help.

closed 2022-11-28 by bblanchon