bblanchon/ArduinoJson #1829: estimation assistant

thiebatgit, 2022-12-05:
**Description**
Bonjour,
j'ai un fichier de sauvegarde qui contient 300 valeurs et tient très facilement dans le spiffs. Lorsque j'utilise arduinojson assistant avec  "periode : travail": "jour",
  "data": [ 1,2,3,4,5,..........280     
la réponse est :This may not fit in the RAM. Make sure there is enough free space.
Ma question : pourquoi le format arduinojson prend autant de place (3072 recommandé) ?
Alors que le fichier.txt séparé par des & fait 2ko.

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns serialization
3. Output is incomplete
4. Program doesn't call <code>serializeJson()</code> in a loop
5. <code>JsonDocument::overflowed()</code> returns <code>true</code>
6. Increasing the capacity of the <code>JsonDocument</code> doesn't solve the issue.

**Environment**
* Microcontroller: esp8266 wemos d1 mini
* Core/Framework: esp8266 2.7.4
* IDE: 1.8.19

**Reproduction code**
```c++
Data structures	2256	Bytes needed to stores the JSON objects and arrays in memory 
Strings	5	Bytes needed to stores the strings in memory 
Total (minimum)	2261	Minimum capacity for the JsonDocument.
Total (recommended)	3072	Including some slack in case the strings change, and rounded to a power of two
 This may not fit in the RAM. Make sure there is enough free space.
```

bblanchon, 2022-12-08:
Hi @thiebatgit,

Please apologize for the late answer, but I wanted to give you a comprehensive response.
Here it is: https://arduinojson.org/v6/faq/why-does-the-document-take-up-so-much-ram/

Best regards,
Benoit

thiebatgit, 2022-12-08:
Bonjour Benoît,
Merci beaucoup d'avoir pris le temps de me répondre. Je vais étudier votre
réponse avec grand intérêt.
Cordialement.

closed 2022-12-26 by bblanchon