bblanchon/ArduinoJson #1833: Add JSON validation only?

Jeroen88, 2022-12-12:
**Is your feature request related to a problem? Please describe.**  
Not a problem. just a useful function..

**Describe the solution you'd like**  
A function ```DeserializationError error = deserializeJson(json)``` that just reports if the input JSON is a valid JSON.

**Describe alternatives you've considered**  
I do:
```
  StaticJsonDocument<0> emptyDoc;
  StaticJsonDocument<0> filter;

  DeserializationError error = deserializeJson(emptyDoc, json, DeserializationOption::Filter(filter));
```
and that does exactly what I want. But I depend on undefined semantics: keys not present in the filter might as well be skipped instead of parsed, but it seems they are parsed too.

**Additional context**  
None. It is not a big deal, the above alternative is also OK.

Jeroen88, 2022-12-13:
Hi Benoit,

Thank you for your excellent answers! That is really appreciated. Sorry for not checking all the issues, although I did a search for similar questions. 

Also thanks for the explanation of how the parsing is done with an empty filter, now I know what to expect of the call: not perfect but good enough.

It would be nice if you add the suggestion to the "How to" section!

Best regards, Jeroen

bblanchon, 2022-12-15:
Here is the new page: [How to validate a JSON document?](https://arduinojson.org/v6/how-to/validate-json/)

Jeroen88, 2022-12-15:
Very nice and thanks for the reference! One typo in the text: 

> it will oversee**s** some errors in the input

HamzaHajeir, 2023-08-17:
Hi there, I was to raise issue but found commenting here is better

I have a problem in validating Json, as it returns true always:

MVCE (:

```cpp
#include "ArduinoJson.h"

bool validateJson(const char* input) {
    StaticJsonDocument<0> doc, filter;
    return deserializeJson(doc, input, DeserializationOption::Filter(filter)) == DeserializationError::Ok;
}
void setup(){

    char data[] = "HelloWorld";
    Serial.printf("validateJson(%s) --> %d\n", data, validateJson(data));
}
```

Output:  
> validateJson(HelloWorld) --> 1

Tested with ArduinoJson version 6.19.4

HamzaHajeir, 2023-09-11:
No problem @bblanchon, I'm wondering whether there's an effective way of checking the validity of a JSON object around?

Best,

bblanchon, 2023-09-13:
If the `validateJson()` is too forgiving for your needs, you'll have to do full deserialization (i.e., without filter) and check the error code.
Again, I don't plan to add any validation feature in ArduinoJson.

closed 2023-09-13 by bblanchon