bblanchon/ArduinoJson #1840: Custom converter support for some STL containers such as std::vector, std::array, std::map, etc...

laferrierepascal, 2022-12-23:
So this might be something that's already available and I just haven't been able to figure out the right combination of syntax to use in order to get it to work, but I've been trying to implement some custom converters for nested custom types, some of which are simply typedefs of STL containers (namely std::array) that contain one of my custom types. 

Below is a minimal example of what I'm trying to achieve that results in compilation errors.

```
constexpr uint8_t POINT_LIST_SIZE = 10;

typedef struct Point
{
    Point() : x(.0), y(.0) {};

    double x, y;

} Point;

typedef  std::array<Point, POINT_LIST_SIZE> PointList;

void convertFromJson(JsonVariantConst src, Point& p) 
{
    p.x = src["x"];
    p.y = src["y"];
}

void convertFromJson(JsonVariantConst src, PointList& list) 
{
    size_t index = 0;
    size_t maximumIndex = list.size();

    if (maximumIndex > 0)
    {
        for (auto p : src.as<JsonArrayConst>())
        {
            list[index++] =  p.as<Point>();

            if (index >= maximumIndex) { break;}
        }
    }
}
```

Some of the errors that I'm getting are:
error: 'bool canConvertFromJson(ArduinoJson::JsonVariantConst, const PointList&)' redeclared as different kind of entity

ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:32:30: error: 'canConvertFromJson' cannot be used as a function
return canConvertFromJson(src, dummy);  // Error here? See https://arduinojson.org/v6/unsupported-is/

The reason I believe this is happening is because the template functions only look at the type that's passed in and don't have any mechanism to detect STL containers and redirect to generic algorithms to parse the contents as arrays. 

So far I've been able to get the results I need by explicitly defining another struct and making slight adjustments to the conversion function:
```
typedef struct PointList 
{
    PointList() {};

    std::array<Point, POINT_LIST_SIZE> points;
}
PointList;

void convertFromJson(JsonVariantConst src, PointList& list) 
{
    size_t index = 0;
    size_t maximumIndex = list.points.size();

    if (maximumIndex > 0)
    {
        for (auto p : src.as<JsonArrayConst>())
        {
            list.points[index++] =  p.as<Point>();
            if (index >= maximumIndex) { break;}
        }
    }
}
```

Ideally I'd like to be able to forego having to redo much of the boilerplate code involved in creating a new class to wrap a container. This was an extremely useful feature that I was able to use in another C++ JSON library (nlohmann::json) which I've had to ditch due to now working on an embedded device. Depending on the size of the project, the number of explicit structure definitions for any custom type within different STL containers can grow rapidly.

Again, I'm not sure if this is currently possible so if there's a way to achieve this I'd be really greatful for some help in getting this working.

Thank you.

bblanchon, 2022-12-23:
Hi @laferrierepascal,

I added the missing page on the site; sorry about that.
I'll add the converters for STL containers in the how-to section as soon as possible.

Best regards,
Benoit

bblanchon, 2022-12-24:
[How to create converters for STL containers?](https://arduinojson.org/v6/how-to/create-converters-for-stl-containers/)

laferrierepascal, 2022-12-29:
Hi @bblanchon ,

Thanks for the quick reply, especially over the holidays. I've managed to get the custom converter working for the std::vector thanks to your example, but I've been struggling (as I am far from a TMP wizard) with trying to get one working for std::array mainly due to the 2nd template parameter for specifying the compile time size of the array. This may be a case of me trying to eagerly optimize my code to avoid heap fragmentation down the road since I'm working on an embedded device. I tried browsing the source code (namely ConverterImpl.hpp and to see if I could discern some sort of pattern to apply the enable_if_t tricks you alluded to but came up short. I'd hate to ask for yet another specific example, but I've been at it for a few days now and I'm getting dizzy hehe.

bblanchon, 2022-12-29:
Hi @laferrierepascal,

Sorry, I've put you on the wrong track for `std::array`: we cannot use the same implementation because `std::array` doesn't support the `psuh_back()` operation.
I added the converter for `std::array` in the how-to.

Best regards,
Benoit

laferrierepascal, 2022-12-30:
Hi @bblanchon,

Thanks again for all the hard work, this is perfect for my uses. Hopefully others will be able to benefit from this as well.

Happy coding!

KSDaemon, 2023-01-28:
Hi @bblanchon Thnx for the great page in the docs! 
I have a question about recursion. Let's say I have a map of maps: 
```cpp
typedef std::unordered_map<const char *, std::unordered_map<const char *, bool>> ClientRoles;
```

I created a templated converted like this:
```cpp
template<typename T>
struct Converter<std::unordered_map<const char *, T> > {
    static void toJson(const std::unordered_map<const char *, T> &src, JsonVariant dst) {
        JsonObject obj = dst.to<JsonObject>();
        for (const auto& item : src)
            obj[item.first] = item.second;
    }

    static std::unordered_map<const char *, T> fromJson(JsonVariantConst src) {
        std::unordered_map<const char *, T> dst;
        for (JsonPairConst item: src.as<JsonObjectConst>())
            dst[item.key()] = item.value();
        return dst;
    }

    static bool checkJson(JsonVariantConst src) {
        JsonObjectConst obj = src;
        bool result = obj;
        for (JsonPairConst item: obj)
            result &= item.value().is<T>();
        return result;
    }
};
```

While it works for maps with simple values, there are problems with complex types on the `second` side.

At first place I was thinking that it is enough to deal with the recursion. For example, during deserialization
`dst[item.key()] = item.value();` will see that `item.value()` is not a simple value but some more complex type, try to search and find the same converter. But the compiler says that it can't find.

The code: 
```cpp
m_routerRoles = doc[2].as<ClientRoles>();
```

Complains: 
```sh
In file included from /src/WampCaster.h:20:0,
                 from /WampCaster-Test/WampCaster-Test.ino:4:
/src/TypeConverters.h: In instantiation of 'static std::unordered_map<const char*, T> ArduinoJson6200_F1::Converter<std::unordered_map<const char*, T> >::fromJson(ArduinoJson6200_F1::JsonVariantConst) [with T = std::unordered_map<const char*, std::unordered_map<const char*, bool> >]':
/ArduinoJson/Variant/VariantRefBase.hpp:54:34:   required from 'typename ArduinoJson6200_F1::enable_if<(((! ArduinoJson6200_F1::is_same<T, char*>::value) && (! ArduinoJson6200_F1::is_same<T, char>::value)) && (! ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value)), T>::type ArduinoJson6200_F1::VariantRefBase<TDerived>::as() const [with T = std::unordered_map<const char*, std::unordered_map<const char*, std::unordered_map<const char*, bool> > >; TDerived = ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::JsonDocument&>; typename ArduinoJson6200_F1::enable_if<(((! ArduinoJson6200_F1::is_same<T, char*>::value) && (! ArduinoJson6200_F1::is_same<T, char>::value)) && (! ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value)), T>::type = std::unordered_map<const char*, std::unordered_map<const char*, std::unordered_map<const char*, bool> > >]'
/src/WampCaster-Impl.h:363:44:   required from here
/src/TypeConverters.h:151:16: error: no match for 'operator[]' (operand types are 'std::unordered_map<const char*, std::unordered_map<const char*, std::unordered_map<const char*, bool> > >' and 'ArduinoJson6200_F1::JsonString')
             dst[item.key()] = item.value();
                ^
In file included from /Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/unordered_map:48:0,
                 from /src/Constants.h:25,
                 from /src/WampCaster.h:18,
                 from /WampCaster-Test/WampCaster-Test.ino:4:
/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/unordered_map.h:667:7: note: candidate: std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator[](const key_type&) [with _Key = const char*; _Tp = std::unordered_map<const char*, std::unordered_map<const char*, bool> >; _Hash = std::hash<const char*>; _Pred = std::equal_to<const char*>; _Alloc = std::allocator<std::pair<const char* const, std::unordered_map<const char*, std::unordered_map<const char*, bool> > > >; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type = std::unordered_map<const char*, std::unordered_map<const char*, bool> >; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::key_type = const char*]
       operator[](const key_type& __k)
       ^
/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/unordered_map.h:667:7: note:   no known conversion for argument 1 from 'ArduinoJson6200_F1::JsonString' to 'const char* const&'
/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/unordered_map.h:671:7: note: candidate: std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator[](std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::key_type&&) [with _Key = const char*; _Tp = std::unordered_map<const char*, std::unordered_map<const char*, bool> >; _Hash = std::hash<const char*>; _Pred = std::equal_to<const char*>; _Alloc = std::allocator<std::pair<const char* const, std::unordered_map<const char*, std::unordered_map<const char*, bool> > > >; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type = std::unordered_map<const char*, std::unordered_map<const char*, bool> >; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::key_type = const char*]
       operator[](key_type&& __k)
       ^
/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/unordered_map.h:671:7: note:   no known conversion for argument 1 from 'ArduinoJson6200_F1::JsonString' to 'const char*&&'

```

So I'm definitely missing some knowledge or understanding of how it works. Or just the code in my converter is wrong...

bblanchon, 2023-01-29:
> error: no match for 'operator[]' (operand types are 'std::unordered_map<...>' and 'JsonString')

You forgot to call [`JsonString::c_str()`](https://arduinojson.org/v6/api/jsonstring/), like so:

```c++
dst[item.key().c_str()] = item.value();
```

@KSDaemon, with your permission, I'd like to include your converter class in [How to create converters for STL containers?](https://arduinojson.org/v6/how-to/create-converters-for-stl-containers/)

KSDaemon, 2023-01-29:
@bblanchon O thnx! I thought I'd missed some little thing!

> I'd like to include your converter class in [How to create converters for STL containers?](https://arduinojson.org/v6/how-to/create-converters-for-stl-containers/)

Sure! If that can be helpful to others — I'm all for it!

Here are updated versions of converters that cover most of the JSON-applicable types. Feel free to pick any or all of them. 
```cpp
#include <unordered_map>
#include <map>
#include <vector>
#include <list>
#include <ArduinoJson.h>

namespace ARDUINOJSON_NAMESPACE {

// Converter for templated std::vector
template<typename T>
struct Converter<std::vector<T> > {
    static void toJson(const std::vector<T> &src, JsonVariant dst) {
        JsonArray array = dst.to<JsonArray>();
        for (T item: src)
            array.add(item);
    }

    static std::vector<T> fromJson(JsonVariantConst src) {
        std::vector<T> dst;
        for (T item: src.as<JsonArrayConst>())
            dst.push_back(item);
        return dst;
    }

    static bool checkJson(JsonVariantConst src) {
        JsonArrayConst array = src;
        bool result = array;
        for (JsonVariantConst item: array)
            result &= item.is<T>();
        return result;
    }
};

// Converter for templated std::list
template<typename T>
struct Converter<std::list<T> > {
    static void toJson(const std::list<T> &src, JsonVariant dst) {
        JsonArray array = dst.to<JsonArray>();
        for (T item: src)
            array.add(item);
    }

    static std::list<T> fromJson(JsonVariantConst src) {
        std::list<T> dst;
        for (T item: src.as<JsonArrayConst>())
            dst.push_back(item);
        return dst;
    }

    static bool checkJson(JsonVariantConst src) {
        JsonArrayConst array = src;
        bool result = array;
        for (JsonVariantConst item: array)
            result &= item.is<T>();
        return result;
    }
};

// Converters for templated string-keys std::map
template<typename T>
struct Converter<std::map<const char *, T> > {
    static void toJson(const std::map<const char *, T> &src, JsonVariant dst) {
[... 124 lines cut ...]
    }

    static std::pair<std::string, T> fromJson(JsonVariantConst src) {
        std::pair<std::string, T> dst;
        for (JsonPairConst item: src.as<JsonObjectConst>())
            dst = std::make_pair(item.key().c_str(), item.value().as<T>());
        return dst;
    }

    static bool checkJson(JsonVariantConst src) {
        JsonObjectConst obj = src;
        bool result = obj;
        if (obj.size() != 1) return false;
        for (JsonPairConst item: obj)
            result &= item.value().is<T>();
        return result;
    }
};

}  // namespace ARDUINOJSON_NAMESPACE
```

One of the last missing pieces is the converter for `std:tuple`. But well... It is not so easy %) because you can not access tuple elements via index as a variable in runtime. and if you target c++11 — then you even do not have a `std::apply`, `std::make_sequence`. So I decided to save my brains for now :)

If someone needs tuple support — he can always write a converter for a tuple of 1−2−3 items. I don't think storing a tuple with more than 3−4 items is a good idea.

closed 2023-02-08 by bblanchon