bblanchon/ArduinoJson #1841: ESP32-S3 Deserialization of JSON from SPIFFS crash

cyberdevnet, 2022-12-28:
**Describe the issue**  
I'm trying to deserialize a json file which is an array of objects containg about ~500 entries but the board crashes

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. The program crashes
4. Program crashes before calling <code>deserializeJson()</code>
5. Program doesn't use <code>StaticJsonDocument</code>
6. Reducing stack usage doesn't prevent the crash

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP-ROM:esp32s3-20210327 Build:Mar 27 2021
* IDE: [VScode]

**Reproduction**  
JSON: 
```
[
{ "id": 1, "name": "Michigan", "pgm": "1","cc1": "1","cc2": "1","cc3": "1" },
{ "id": 2, "name": "Del Sol", "pgm": "2","cc1": "2","cc2": "2","cc3": "2" },
{ "id": 3, "name": "Scoville", "pgm": "3","cc1": "3","cc2": "3","cc3": "3" },
{ "id": 4, "name": "Moulton", "pgm": "4","cc1": "4","cc2": "4","cc3": "4" },
{ "id": 5, "name": "Sauthoff", "pgm": "5","cc1": "5","cc2": "5","cc3": "5" }

// 500 more entries.....
]

// I'm opening and deserializing the file with the following function:

DynamicJsonDocument openJsonPresets(String fileName)
{
    File f = SPIFFS.open(fileName, "r");
    const size_t capacity = ESP.getMaxAllocHeap();

    DynamicJsonDocument doc(capacity);
    Serial.println("capacity...");
    Serial.println(doc.capacity());   // prints 229364

    DeserializationError err = deserializeJson(doc, f);
    if (err == DeserializationError::Ok)
        doc.shrinkToFit();
    else
        Serial.println("error...");
    Serial.println(err.c_str());
    f.close();
    return doc;
}

// and just trying to print some entries with:

void setPresetData()
{

    DynamicJsonDocument doc = openJsonPresets("/preset_data3.json");
    for (JsonObject item : doc.as<JsonArray>())
    {

        const char *thename = item["name"]; // "Michigan", "Del Sol", "Scoville", "Moulton", "Sauthoff", ...
        const char *thepgm = item["pgm"];
        const char *thecc3 = item["cc3"];

        Serial.print("name ");
        Serial.println(thename);
        Serial.print("pgm ");
        Serial.println(thepgm);
        Serial.print("cc3 ");
        Serial.println(thecc3);
    }
};

```

**Program output**  

Actual output:

```
capacity...
229364
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK) 
Core  0 register dump:
PC      : 0x403804b3  PS      : 0x00060036  A0      : 0x8037f3d0  A1      : 0x3fced290  
A2      : 0x3fc962d0  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x3fc95c10  A12     : 0x00000000  A13     : 0x3fc99dfc  
A14     : 0x02c962d0  A15     : 0x00ffffff  SAR     : 0x0000000e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace:0x403804b0:0x3fced2900x4037f3cd:0x3fced2d0 0x4037dcec:0x3fced300 0x4037dce2:0xa5a5a5a5  |<-CORRUPTED
```
Sorry if this seems a stupid issue, please understand I'm pretty new with Arduino/C++/ESP32...

cyberdevnet, 2022-12-28:
the file I'm trying to open seems too big and it's causing the crash, if I reduce the entries from 500 to ~300 it does not crash.
Is there anything I can do to handle all those object?

bblanchon, 2022-12-29:
Hi @cyberdevnet,

The `JsonDocument` may be so big that it doesn't leave enough room for anything else.
Please try to reduce its size.
You can use the ArduinoJson Assistant to determine the right capacity.

Best regards,
Benoit

cyberdevnet, 2022-12-29:
ArduinoJson Assistant says 65560 and I tried also that value.
I switched to SpiRamJsonDocument since I have an ESP32 with 8MB og PSRAM but still the same issue. I'm getting crazy...

PSRAM: 
  Total: 8386083
  Used: 692
  Free: 8385391
  Largest block: 8257524
  Minimum free since boot: 8385319

cyberdevnet, 2022-12-29:
I'm not sure if PSRAM is being used, I configured the following:

`
struct SpiRamAllocator
{
    void *allocate(size_t size)
    {
        return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
    }
    void deallocate(void *pointer)
    {
        heap_caps_free(pointer);
    }
    void *reallocate(void *ptr, size_t new_size)
    {
        return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);
    }
};

using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
`
and used like this:

`    File file = SPIFFS.open("/preset_data128.json", FILE_READ);

    SpiRamJsonDocument Namefilter(200);

    JsonObject filter_0 = Namefilter.createNestedObject();
    filter_0["id"] = true;
    filter_0["name"] = true;

    SpiRamJsonDocument doc(300000);
    DeserializationError error = deserializeJson(doc, file, DeserializationOption::Filter(Namefilter));`

Even with filter is crashing.
sorry for the mess in formatting :(

cyberdevnet, 2022-12-29:
ok PSRAM is being used but it crashes during serialization, this value I took just before serialization:
`SpiRamJsonDocument doc(300000);
    Serial.println("Heap doc: ");
    Serial.printf("  Total: %u\n", ESP.getHeapSize());
    Serial.printf("  Used: %u\n", ESP.getHeapSize() - ESP.getFreeHeap());
    Serial.printf("  Free: %u\n", ESP.getFreeHeap());
    Serial.printf("  Largest block: %u\n", ESP.getMaxAllocHeap());
    Serial.printf("  Minimum free since boot: %u\n", ESP.getMinFreeHeap());

    Serial.println("PSRAM doc: ");
    Serial.printf("  Total: %u\n", ESP.getPsramSize());
    Serial.printf("  Used: %u\n", ESP.getPsramSize() - ESP.getFreePsram());
    Serial.printf("  Free: %u\n", ESP.getFreePsram());
    Serial.printf("  Largest block: %u\n", ESP.getMaxAllocPsram());
    Serial.printf("  Minimum free since boot: %u\n", ESP.getMinFreePsram());
    DeserializationError error = deserializeJson(doc, file, DeserializationOption::Filter(Namefilter));`

here the result:
`Heap doc: 
  Total: 334580
  Used: 112684
  Free: 221896
  Largest block: 212980
  Minimum free since boot: 221852
PSRAM doc: 
  Total: 8386051
  Used: 300892
  Free: 8085159
  Largest block: 7995380
  Minimum free since boot: 8085159`

as you can see is using the "300000" assigned but during serialzation it crashes:
`Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK) 
Core  0 register dump:
PC      : 0x40381f9f  PS      : 0x00060036  A0      : 0x80380be0  A1      : 0x3fca71d0  
A2      : 0x3fc98870  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x3fc98844  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x3fca72a0  
A14     : 0x02c98870  A15     : 0x00ffffff  SAR     : 0x0000001c  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000  

Backtrace: 0x40381f9c:0x3fca71d0 0x40380bdd:0x3fca7210 0x4037f4bc:0x3fca7240 0x4037f4b2:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: b7f770e40b35db8c`

bblanchon, 2022-12-30:
Your original message said you had an issue with *deserialization*, is this solved?
Can you try to run the Troubleshooter again?

cyberdevnet, 2022-12-30:
Hi, nope but I think the issue is caused by the BLE stack since I'm calling the function via BLE characteristic. When I deserialize at setup() everything is ok.
Calling deserialize via BLE characteristic probably causes the memory consuption and the crash, even with really small json.
The Stack goes from 5744 to 500, below that value it crashes.
I think you can close this issue. THX

bblanchon, 2022-12-31:
Indeed, you may not be allowed to perform long operations in a BLE callback.
You should put the message in a queue and process it in the main loop.

closed 2022-12-31 by bblanchon