bblanchon/ArduinoJson #1842: Arduino Json v6.20.0 is much slower than v6.19.4 on ESP8266

sblantipodi, 2022-12-28:
**Describe the bug**  

Hi, since I upgraded Arduino Json to v.6.20.0 I immediately noticed a huge performance degradation on my firmware.
https://github.com/sblantipodi/glow_worm_luciferin

The firmware basically manages RGB LEDs to act like an ambilight.
Color informaion is sent via UDP stream and does not even use Arduino JSON for it, this is very weird...

When using v6.20.0 I can't process more than 40 frames per second, 
If I revert back to v6.19.4 performance is good as always and goes over 60 frames per second no problem.

I sincerely don't have any idea on what can cause it...

Any idea ? Am I the only one who is noticing performance degradation on v6.20.0 on ESP8266?

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266
* Core/runtime: PlatformIO espressif8266@4.0.1
* IDE: CLion

sblantipodi, 2022-12-28:
the strange thing is that there are something that is lowering performances that I can't understand...

If I send 60 frame to the ESP, it process 40 frames,
If I send 30frame to the ESP, it process 20 frames,
If I send 15 frame to the ESP, it process 8 frames.

So weird...

bblanchon, 2022-12-29:
Hi @sblantipodi,

Can you point me to the line of code where you experience the slowdown?

Best regards,
Benoit

sblantipodi, 2022-12-29:
Hi @bblanchon, thank you very much for the answer.

https://github.com/sblantipodi/glow_worm_luciferin/blob/master/src/NetworkManager.cpp#L33

I sincerely don't understand how ArduinoJson can be related to this problem,
if I revert back to v6.19.4 it's an instant fix.

bblanchon, 2022-12-29:
Where do you call ArduinoJson?

sblantipodi, 2022-12-29:
> Where do you call ArduinoJson?

there is no call to ArduinoJson while processing the UDP stream (when there is performance degradation)... 
this is the weird thing... 

ArduinoJson is widely used in other part of the software example:
https://github.com/sblantipodi/glow_worm_luciferin/blob/master/src/NetworkManager.cpp#L702

but that part of the code is not "hitted" while processing the UDP stream...

bblanchon, 2022-12-29:
I don't see anything that could explain the slowdown.
Could you do a `git bisect` to pinpoint the commit causing the issue?

sblantipodi, 2023-01-10:
Hi, I'm pretty sure that the problem isn't something related to Arduino JSON. 
I was able to reproduce the problem by changing some lines of codes that should not trigger the problem.

It's a very weird problem but it's not something related to the mighty Arduino JSON.

Sorry for the false issue and thanks for the answer.

closed 2023-01-10 by sblantipodi

bblanchon, 2023-01-11:
> I was able to reproduce the problem by changing some lines of codes that should not trigger the problem.

This sure sounds like an undefined behavior.
However, UBs usually don't slow down the program; they just make it crash.