bblanchon/ArduinoJson #1843: Overwrite string values (non-const char *) of the same length instead of allocating them into the Json doc?

Jeroen88, 2022-12-28:
**Is your feature request related to a problem? Please describe.**  
I use a DynamicJsonDocument to frequently store c strings (not const) and these strings are always of the same length. If I use ```doc["key"] = value;``` repeatedly, the document is overflowed  after a while (I am aware of the information in "I found a memory leak in the library!")

**Describe the solution you'd like**  
Would it be possible to _overwrite_ c strings of the same length (or even strings that are shorter) at the same key instead of _allocating_ them into the Json document?

**Describe alternatives you've considered**  
If I do this:
```
JsonVariant test = doc["key"];
// Make sure the object at 'key' is already present and it's value is equal or longer than the new value
if(!test.isNull() && strlen(test.as<const char *>() >= the lengh of the new value) {  
  const char * ptr = test.as<const char *>();
  strcpy((char *) ptr, the addess of the new value, the length of the new value);
} else {
  // Do a regular insertion into the Json document
  doc["key"] = the new value;
}
```
This perfectly works  (my c strings are dynamically created and get out of scope, so I can not use the 'const char *' feature: they need to be copied into the Json document).

**Additional context**  
Although I am not sure, a ```#define``` like is used [here](https://github.com/bblanchon/ArduinoJson/blob/5094b84a46546b31c758a621d6346aea1d32de27/src/ArduinoJson/Memory/MemoryPool.hpp#L77) might optionally add such a feature? But maybe it is impossible to add this to the _monotonic allocator_?

bblanchon, 2022-12-29:
Hi @Jeroen88,

Since ArduinoJson 6.16, the `JsonDocument` stores only one copy of each string.
As a consequence, string instances are shared, so overriding the characters may have unexpected results if you use this string multiple times.
To keep the code as small as possible, I didn't include a reference counter, so the `JsonDocument` doesn't know if the string is actually shared.
I don't plan on adding such a complex feature until v7. Sorry.

As a workaround, I suggest you:

* pass a `const char*` so that the string is stored by the pointer
* recreate the `JsonDocument` on each serialization 

Best regards,
Benoit

bblanchon, 2023-07-18:
In branch [7.x](https://github.com/bblanchon/ArduinoJson/tree/7.x), strings are ref-counted, which fixes this issue.

closed 2023-07-18 by bblanchon

Jeroen88, 2023-08-05:
> In branch [7.x](https://github.com/bblanchon/ArduinoJson/tree/7.x), strings are ref-counted, which fixes this issue.

Thnx Benoit!