bblanchon/ArduinoJson #1844: ArduinoJson as an ESP-IDF package

plasmapper, 2022-12-30:
Dear Benoît,

ESP-IDF has a [component registry](https://components.espressif.com/) that already allows developers to sign-in for preview to release a component. Would you please consider releasing ArduinoJson as an ESP-IDF package. Using it as a package will simplify the installation and versioning for those who use ESP-IDF as it already does for those developing with Arduino IDE.

Thank you.

bblanchon, 2022-12-30:
Hi @plasmapper,

I requested access to the closed preview.
I'll keep you informed of the progress.

Best regards,
Benoit

bblanchon, 2023-01-13:
@plasmapper, ArduinoJson is now in the registry:
https://components.espressif.com/components/bblanchon/arduinojson
Please give it a try.

plasmapper, 2023-01-13:
Great, thanks!

commit b84afd31 (2023-01-13): Add to IDF Component Registry (closes #1844)
commit 78851556 (2023-01-14): Add to IDF Component Registry (closes #1844)
closed 2023-01-13 by plasmapper