bblanchon/ArduinoJson #1846: containsKey

sevelm, 2023-01-05:
**Describe the issue**  
Using containsKey. It is recommended not to use this. in the instructions we talk about using this version instead:

`
const char* error = root["error"];
if (error) {
  Serial.println(error);
  return;
}
`
**Troubleshooter report**  
not all data type can be retrieved like this, for example int.

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* IDE: Arduino IDE 1.8.19

**Reproduction** 
this should be avoided:
`
if (doc.containsKey("ChContr4_contr")) { mb.Hreg(Device[websDevNbr].NbrTCPwriteReg+2, doc["ChContr4_contr"]); }
`
**Question** 
Is there a way to find out whether the JSON has a corresponding key independently of the data type.

bblanchon, 2023-01-06:
Hi @sevelm,

You can use [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) or [`JsonVariantConst`](https://arduinojson.org/v6/api/jsonvariantconst/):

```c++
JsonVariantConst value = doc["key"];
if (value != nullptr) {
  Serial.println(value.as<int>());
}
```

Best regards,
Benoit

sevelm, 2023-01-06:
JsonVariantConst val;
        `
        val = doc["brand"]; if (val != nullptr) { Serial.println(val.as<String>()); }
        val = doc["wallbox"]; if (val != nullptr) { Serial.println(val.as<int>()); }
        ..... and many more queries
`

bblanchon, 2023-01-09:
I don't think I'll repeat this pattern a hundred times, but it really depends on the use case.

You can make the test a bit safer by checking the type explicitly with [`is<T>()`](https://arduinojson.org/v6/api/jsonvariant/is/) instead of comparing with `nullptr`.

Also, if you're going to do the test-type-and-print operation repeatedly, you should create a generic function, like so:

```c++
void checkTypeAndPrint
(

closed 2023-01-09 by bblanchon