bblanchon/ArduinoJson #1852: LoadProhibited exception on ESP32

AcuarioCat, 2023-01-09:
My program creates a json every second to send to a web browser.
Several times a day (sometimes 2, sometimes more) I get a LoadProhibited exception and reboot.
I never have a day when there aren't at least 2 reboots.

It's impossible to write a code snippet to replicate this as it is such an infrequent occurrence.

The cause (from the stack dump) is always in the textformatter.hpp writeString routine with EXCVADDR = 0. The Expressif documentation points to the most likely cause being an attempt to dereference a NULL pointer.
I have modified the writeString code to the following to test for this (the existing code should but seems to fail sometimes??):

     while (*value)
      {
          writeChar(*value);
          if (*(value + 1) == NULL)
              break;
          *value++;
      }

..and in the last 24 hours I have not had a single reboot.
I have not modified anything else in my code.
I will update if I get any more exceptions but at the moment it seems a fix.

bblanchon, 2023-01-09:
Hi @AcuarioCat,

In summary, you replaced:

```c++
while (*value)
      writeChar(*value++);
```

with 

```c++
while (*value) {
    writeChar(*value);
    if (*(value + 1) == NULL)
        break;
    *value++;
}
```

These two snippets should do exactly the same thing; your exception is probably due to an undefined behavior.
I bet the `value` pointer is dangling; please double-check.

Please also include the Troubleshooter report so I can add a scenario matching your issue.

Best regards,
Benoit

AcuarioCat, 2023-01-14:
I have changed from using a StaticJsonDocument to a DynamicJsonDocument and at the moment the problem seems to have disappeared.

closed 2023-01-14 by bblanchon