bblanchon/ArduinoJson #1858: Unused variables causes compile error

matswebjorn, 2023-01-16:
**Description**
In extras/tests/JsonVariant/containsKey.cpp and extras/tests/JsonVariant/remove.cpp is
static const char* null = 0;
defined but not used. This causes compiler to abort with unused variable error message

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: Raspberry Pi Zero 2 W
* Core/Framework: Buildroot 2022.11
* IDE: Buildroot 2022.11 SDK

bblanchon, 2023-01-17:
Hi @matswebjorn,

Thank you for reporting this.
Indeed, the variable is unused and should be removed, but before fixing it, I'd like to trigger the same error on the CI.
I tried to add `-Wunused-variable`, but it did report the error.
Can you copy/paste the complete error line?

Best regards,
Benoit

bblanchon, 2023-02-03:
@matswebjorn?

bblanchon, 2023-02-09:
I finally managed to reproduce the issue after upgrading Catch.
Indeed, Catch 1 defined the following pragma:

```c++
#pragma GCC diagnostic ignored "-Wunused-variable"
```

This explains why I could not get this error 😄

commit 38dda5e6 (2023-02-09): Remove unused variables (fixes #1858)
commit 78280922 (2023-02-09): Remove unused variables (fixes #1858)
commit 034a3f72 (2023-02-10): Remove unused variables (fixes #1858)
commit 4abf0302 (2023-02-10): Remove unused variables (fixes #1858)
closed 2023-02-10 by bblanchon