bblanchon/ArduinoJson #1860: Issue with JsonVariant.as<T>() where T is char *  Need to update docs!

djaus2, 2023-01-18:
**Describe the bug**  
When using the example to iterate thjrough elements of a Json Document ins the part ```.as<T>()``` can't use char* for T

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at compile time
2. Error says &quot;no matching function for call to <code>...(&lt;unresolved overloaded function type&gt;)</code>&quot;
3. Adding the parentheses doesn't fix the issue .. Weren't missing in the parenthesis first place.

PS Liked that trouble shooter.

**Environment**  
Here is the environment that I used:
* Microcontroller: RP2040
* RPI Pico W
* IDE: Arduino IDE 1.2.03

**Reproduction**  
This is a compile time error only.
Small sketch with same code as per documentation:
```
  DynamicJsonDocument doc(1024);
  deserializeJson(doc, "{\"first\":\"\",\"second\":\"second\"}");
  JsonObject root = doc.as<JsonObject>();

  // using C++11 syntax (preferred):
  for (JsonPair kv : root) {
      Serial.println(kv.key().c_str());
      Serial.println(kv.value().as<char *>());
  }
```

Workaround: replace char * with String

Further: I see its deprecated anyway
const char*        as<char*>() const;  // ⚠️ deprecated since 6.18
const char*        as<const char*>() const;

SO DOCUMENTATION NEEDS UPDATE.. I spent hours on this!
But thanks anyway for the library.

**Compiler output**    
Compilation error: no matching function for call to 'ArduinoJson6200_F1::JsonVariant::as<char*>()'

bblanchon, 2023-01-18:
Hi @djaus2,

I'm sincerely sorry you lost so much time on this issue.
I wish I had done a search to fix all the mentions of `as<char*>`

Anyway, the documentation should be up to date now.
I also added a new scenario in the Troubleshooter and modified the library so that you now get the regular error message for unsupported types.

Hopefully, these measures will help other users find the solution quickly.

Best regards,
Benoit

closed 2023-01-27 by bblanchon

djaus2, 2023-01-27:
All good. Docs are good now.. 
Functionality on this issue is clear now.
Keep up teh good work, thx.