bblanchon/ArduinoJson #1861: as<double>() returns value slightly different than input.

kennetn, 2023-01-18:
Compiled with without changing ARDUINOJSON_USE_DOUBLE so JsonFloat is double
I think the issue is within traits::make_float

**Troubleshooter report**  
Did not have report

**Environment**  
Here is the environment that I used:
Visual studio 2019
Arduino 6.20.0

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
DynamicJsonDocument jsonDoc( 100 );

                DeserializationError error = deserializeJson( jsonDoc, "{\"db\":0.1}" );

                if ( DeserializationError::Ok == error )

                {

                                double db = jsonDoc["db"].as<double>();

                                assert( db == 0.1 );

                                // Will be 0.10000000000000001 ( float )

                }
```

bblanchon, 2023-01-18:
Hi @kennetn,

`0.1` cannot be represented in binary form; what you see is the closest value that can be encoded in binary.
Here is a screen capture of Visual Studio 2022 to convince you:

[image]

See [IEEE754](https://en.wikipedia.org/wiki/IEEE_754) for the details.

Best regards,
Benoit

bblanchon, 2023-01-19:
Indeed, it's a different story for 0.123.
`make_float()` creates the value by doing the following operation:

```c++
123 * 1e-1 * 1e-2
```

As you can see, the infamous `0.1` is here, which is probably why we get a slightly different result.

I don't recall all the details, but I know I spent a lot of time optimizing this piece of code, so I'm not sure I'll be able to fix this error without altering the program footprint and performance.

kennetn, 2023-01-23:
Hi

I did a test of you change and it is working. Thanks
Should I close the issue ?

bblanchon, 2023-01-23:
Please, don't close the issue.
That was just an attempt I made quickly, but I didn't merge it.

I reversed the operation when multiplying by powers of 10: instead of `123 * 1e-1 * 1e-2`, I do `123 * 1e-2 * 1e-1`.
It works for `0.123`, but I don't know if it's better or worse for other floats.
This morning, I checked the implementation of `strtod` in avr libc, and they start with 1e-2 too, so maybe there is a logical reason why we should do that.

bblanchon, 2023-01-24:
I ran some tests today to see if reversing the operation order improves the results, but they don't 😢 
It fixes some numbers like `0.123` but breaks others like `0.003`.
This problem is well documented on the Internet ([this article is a good starting point](https://www.exploringbinary.com/how-strtod-works-and-sometimes-doesnt/)), but the solution will significantly increase the size of the code, so I prefer keeping this imperfect but compact implementation for now.

closed 2023-01-24 by bblanchon

kennetn, 2023-01-24:
Is it possible to add some compile switch to make it optional? For me it cold also be ms specific and use strtod. ?

bblanchon, 2023-01-25:
If I added a compile switch every time someone asked for it, I would have ended up with a Quasimodo-like library with dozens of switches that only one (if any) person uses.
Moreover, I don't want to add a flag that slightly changes the behavior in almost unnoticeable ways.

Besides, you can fix your code by adding some tolerance in your test.
This is a very common problem, and that's why every testing library provides facilities to compare floats.