bblanchon/ArduinoJson #1863: Sometimes changing a variable value in a DynamicJsonDocument doesn't work.

quentinmissler, 2023-01-25:
**Describe the bug**  
Sometimes changing a variable value in a DynamicJsonDocument doesn't work.

The MyData.dataJSON variable is an average that I just reset once a minute to zero. Having from time to time erroneous average results, I realized that the reset was not done. The variable keeps its old value.

The problem occurs about once in a five hundred at random intervals. No RAM issues.
The problem occurs all my devices and on different models of ESP32 chip.

I also noticed that on this same variable the incrementation of the average is not carried out (example the variable is worth 10, I increment by 5 and if I do a Serial.print the value of the variable is always 10. This problems occurs at the same intervals and under the same circumstances as the previous problem.).

**Environment**  
Here is the environment that I used:
* Microcontroller: [e.g. ESP32 platform v5.0.0]
* IDE: [e.g. Arduino IDE 2.0.3]
* ArduinoJson v6.19.14

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
DynamicJsonDocument dataJSON(15000);

MyData.dataJSON["clcE"] = 0;
delay(50);
Serial.println(MyData.dataJSON["clcE"].as<String>());
```

Expected output: "0"

Actual output:
The compiler output is sometimes not "0".

bblanchon, 2023-01-26:
Hi @quentinmissler,

That's most likely a fragmentation issue.
Please share the troubleshooter report.

Best regards,
Benoit

quentinmissler, 2023-01-26:
Hi, what do you mean by a fragmentation problem? A troubleshooter report ?

bblanchon, 2023-01-27:
Hi,

Please read [What is Heap Fragmentation?](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html)
In your case, the heap is probably too fragmented to allow a 15000 bytes allocation.
You can quickly validate this theory by checking [`doc.capacity()`](https://arduinojson.org/v6/api/jsondocument/capacity/).
Unfortunately, ArduinoJson 6 cannot work with a fragmented heap, so you need to improve your init code so it behaves correctly.

The [Arduino Troubleshooter](https://arduinojson.org/v6/troubleshooter/) is a self-diagnostic tool that helps you solve the most frequent issues with the library. In the end, it produces a report summarizing your answers.
I'm surprised you don't know about this tool since its URL was included in the template you used to create the present issue.
In your case, it's unlikely that you'll find the solution in the troubleshooter; however, the report will help me improve the tool by adding more choices at the appropriate locations.

Best regards,
Benoit

closed 2023-02-08 by bblanchon