bblanchon/ArduinoJson #1864: Parse array with the same name

khairullaizatt, 2023-01-26:
**Description**
when i parse "title" , i only get the last "title".  I expect all the 10 string "title"  

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns deserialization
3. `deserializeJson()` returns `Ok`
4. Input buffer is modified
5. Casting the pointer doesn't fix the issue

**Environment**
* Microcontroller: ESP8266
* Core/Framework: Deauther ESP8266 Boards Version 2.7.5
* IDE: Version: 2.0.3 Date: 2022-12-05T09:30:25.331Z CLI Version: 0.29.0 [76251df9]  Copyright © 2023 Arduino SA

**Reproduction code**
```c++
DynamicJsonDocument doc(12228);
  DeserializationError error = deserializeJson(doc, client);
  for (JsonObject item : doc["items"].as<JsonArray>()) {
   if (!error) {  
    title = item["title"].as<String>();
    Serial.print("title: ");
    Serial.println(title);
    
    
   } else {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
   }
```

Serial print

00:36:05.241 -> title: Isu bakar al-Quran: Wisma Putra panggil Kuasa Usaha Sementara Kedutaan Besar Sweden
00:36:05.279 -> title: MH17: Waris keluarga terkorban masih harapkan pembelaan
00:36:05.279 -> title: Selangor teliti pembinaan pusat peranginan atas tanah pertanian
00:36:05.312 -> title: #SuaraMalaysia: Kos barangan makanan terus jadi keresahan utama
00:36:05.312 -> title: Kenyataan tular PM setuju Pemuda Pas tunjuk perasaan di Kedutaan Sweden adalah palsu - JPM
00:36:05.312 -> title: Finland catat kadar kelahiran terendah dalam tempoh 150 tahun
00:36:05.348 -> title: KTMB sedia perkhidmatan tren komuter 24 jam sempena Thaipusam
00:36:05.348 -> title: 'UMNO milik semua, bukan MT sahaja' - Khairy
00:36:05.394 -> title: Banjir Mersing: Penduduk sifatkan terburuk dialami dalam 5 tahun
00:36:05.394 -> title: Ribuan pengguna Facebook, Instagram, WhatsApp di AS alami gangguan

how do i parse each title?

bblanchon, 2023-01-27:
Hi @khairullaizatt,

I don't understand your question: your code seems to work correctly.

Best regards,
Benoit

khairullaizatt, 2023-01-27:
hi @bblanchon ,

for (JsonObject item : doc["items"].as<JsonArray>()) {
const char* title = item["title"];

under title there was 10  array with same name ---> 'title'
how do i pointer  'title' outside function.

bblanchon, 2023-01-30:
You could use an array like so:

```c++
String titles[10];

int index = 0;
for (JsonObject item : doc["items"].as<JsonObject>()) {
  title[index++] = item["title"];
  if (index >= 0) break;
}
```

khairullaizatt, 2023-01-30:
hi @bblanchon thank you for your time. 

`String titles[10];

int index = 0;
for (JsonObject item : doc["items"].as<JsonObject>()) {
  title[index++] = item["title"];
  if (index >= 0) break;
}`

Compilation error: conversion from 'ArduinoJson6194_F1::Pair' to non-scalar type 'ArduinoJson::JsonObject {aka ArduinoJson6194_F1::ObjectRef}' requested

below code working,

`  DynamicJsonDocument doc(12228);
  DeserializationError error = deserializeJson(doc, client);
  const JsonArray items = doc["items"].as<JsonArray>();
  title1 = items[0]["title"].as<String>();
  title2 = items[1]["title"].as<String>();
  title3 = items[2]["title"].as<String>();
  title4 = items[3]["title"].as<String>();
  title5 = items[4]["title"].as<String>();
  title6 = items[5]["title"].as<String>();
  title7 = items[6]["title"].as<String>();
  title8 = items[7]["title"].as<String>();
  title9 = items[8]["title"].as<String>();
  title10 = items[9]["title"].as<String>();
    if (!error) {  
    Serial.print("title1: ");
    Serial.println(title1);
    Serial.print("title2: ");
    Serial.println(title2);
    Serial.print("title3: ");
    Serial.println(title3);
    Serial.print("title4: ");
    Serial.println(title4);
    Serial.print("title5: ");
    Serial.println(title5);
    Serial.print("title6: ");
    Serial.println(title6);
    Serial.print("title7: ");
    Serial.println(title7);
    Serial.print("title8: ");
    Serial.println(title8);
    Serial.print("title9: ");
    Serial.println(title9);
    Serial.print("title10: ");
    Serial.println(title10);
    return;
   } `

problem solved.

closed 2023-02-08 by bblanchon