bblanchon/ArduinoJson #1866: Cant make a json output when the output type is char*

Bernard314, 2023-02-03:
**Describe the issue**  
unable to get the json output while using a char* type as output. if i instead use char type as output i am able to get my json output. my code just crashes.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns serialization
3. Program crashes
4. Program uses `StaticJsonDocument`
5. Switching to `DynamicJsonDocument` doesn't fix the issue
6. Program doesn't use `PROGMEM`
7. Program inserts string pointers into the `JsonDocument`
8. Replacing string pointers with literals doesn't fix the crash
9. Output type is `char*`

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32 WROOM devkitcore for Arduino v3.0.3
* IDE: Arduino IDE 2.0.3

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```
#include "ArduinoJson.h"
const int capacity = JSON_OBJECT_SIZE(3); 
char* output;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

StaticJsonDocument<capacity> doc;
//Also tried DynamicJsonDocument doc(2048);

  doc["value"] = "42"; 
  doc["lat"] = "48.748010"; 
  doc["lon"] = "2.293491";
  serializeJson(doc, output, sizeof(output));
}

void loop() {
  // put your main code here, to run repeatedly:
  // Serial.println(output);
}
```

how do i solve this? what am i missing?

bblanchon, 2023-02-04:
Hi @Bernard314,

`char* output` declares a pointer, but no buffer is associated with it.
As a reminder, a pointer is a variable that stores the address of another variable.
In this case, the pointer contains no addresses (it's null), which causes the crash.

You can solve this issue by replacing `output` with a `char` array:

```c++
char output[256];
```

There, `output` points to a static buffer of 256 bytes.

I invite you to read "the missing C++ course" in [my book](https://arduinojson.org/book/).

Best regards,
Benoit

Bernard314, 2023-02-04:
ahhh yes got it, thank you!

closed 2023-02-05 by bblanchon