bblanchon/ArduinoJson #1867: Compilation error: 'StaticJsonDocument' was not declared in this scope

dakoller, 2023-02-03:
**Description**
When I copy the boiler plate text 
"void setup() {
  StaticJsonDocument<200> doc;
  doc["hello"] = "world";
  serializeJson(doc, Serial);
}"

I expect it to work, but I get the mentioned error.

I miss a proper include statement for arduinoJSON in the beginning, but maybe thats not needed.

**Troubleshooter's report**
1. The issue happens at compile time
2. Error says "`doc` does not name a type"
3. Moving statement to `setup()` doesn't fix the issue

**Environment**
* Microcontroller: ESP8266 NodeMCU 12E

**Reproduction code**
```c++
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_BME280.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

...
...

Adafruit_BME280 bme; // use I2C interface
Adafruit_Sensor *bme_temp = bme.getTemperatureSensor();
Adafruit_Sensor *bme_pressure = bme.getPressureSensor();
Adafruit_Sensor *bme_humidity = bme.getHumiditySensor();

ESP8266WiFiMulti WiFiMulti;

//DynamicJsonDocument doc(1024);
//StaticJsonDocument<96> doc;

void setup() {
...

  StaticJsonDocument<200> doc;
  doc["hello"] = "world";
  serializeJson(doc, Serial);

}
```

bblanchon, 2023-02-04:
Hi @dakoller,

I'm afraid you do need to include `ArduinoJson.h`.

Please add the following line at the top of your program:

```c++
#include <ArduinoJson.h>
```

Best regards,
Benoit

dakoller, 2023-02-05:
Thx @bblanchon ! I was looking for this include-line all the time. Now it works.

closed 2023-02-05 by dakoller