bblanchon/ArduinoJson #1870: JsonDocument.as<const char *> -> Pointer to serialized JSON

gonzabrusco, 2023-02-05:
Hello. Sometimes you need to pass the JSON to a function that expects a const char * array. In that case, with ArduinoJSON you are obligated to create a intermediary char array to serialize the JSON and then pass it to that function. 

I'm not sure if it's possible but.... it would be nice to have a JsonDocument.as<const char *> that gives you back a pointer to the serialized data. That data in fact does not really exists (because in that case it would be the same as having an intermediary buffer), it is serialized dynamically based on the [index] requested. 

I think that this is not really possible. But I also think that you know a lot more C++ than I do! So maybe there's an obscure method of doing this that I don't know about it yet.

Thanks for the great library!

bblanchon, 2023-02-06:
Hi @gonzabrusco,

This feature is already available with `as<String>()` and `as<std::string>()`:

```c++
Serial.println(doc.as<std::string>());
```

Best regards,
Benoit

gonzabrusco, 2023-02-06:
I see. But as<String> creates takes up more memory just to create de String right? I wanted to know if there was a possibility of reutilizing the JsonDocument memory instead of asking for more. Like create de Serialized Array without using extra memory.

bblanchon, 2023-02-07:
The `JsonDocument` doesn't store a copy of the serialized JSON document, so it cannot return a pointer.

closed 2023-02-07 by gonzabrusco