bblanchon/ArduinoJson #1871: bits/error_constants.h missing

igoeco, 2023-02-06:
**Description**
I successfully used the ArduinoJSON library on another machine. When I try to run the code on a different machine (using ArduinoIDE), it first complained  that  ArduinoJSON was not installed on line: 
#include <ArduinoJson.h>
So, I used the "Manage Libraries" tool and successfully installed the latest version 6.20.0.
Now I get an error saying:
c:\users\myusername\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory

Clearly, I do not have the file but I am not sure why?

Here are a few more details:
ArduinoIDE version: 2.0.3
Device: ESP32 WROVER Dev 

Thanks for your help.

PS: I have an identical setup on my other machine except it is Windows 11 and not 10

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32 WROVER
* Core/Framework: ESP32 Core
* IDE: Arduino IDE 2.0.3

**Reproduction code**
```c++
#include <ArduinoJson.h>
```

bblanchon, 2023-02-07:
Hi @igoeco,

It seems to be an issue with the ESP32 core.
https://github.com/espressif/arduino-esp32/issues/6593

If you don't have this file, then it's probably a problem with your installation.
Please try to reinstall the ESP32 core.

Several people mentioned an issue with the path length; maybe you can try to shorten it.

Best regards,
Benoit

JEDtoo, 2023-02-19:
I've figured it out and proved it to my satisfaction.<p>
Executive summary: 
<p>
I was getting the failure. I removed two nodes from my path length and it's FIXED. Your suggestion WAS CORRECT!
<p>
Detail:<p>
The path was somewhat long - It was c:/users/jed/OneDrive/electronics/RunableCode/Arduino-1.8.19 then the included files. 1.8.13 worked like that until OneDrive messed it up by adding extra files when I tried to use the IDE while it was syncing and it duplicated some. I installed 1.8.19 to fix and it was failing with the error above. By moving the folder up directly into OneDrive, I fixed it. (yes, those are the wrong / for Windoze but the damn \ hurts my pinky to type and I'm bilingual anyway).
I don't know where the limitation is, but it isn't windows because I've found failures reported under Linux AND I changed my W10 registry to allow longer paths AND I also tested on W11 (easy because on OneDrive on both desktops and my laptop) and it failed everywhere UNTIL I copied from OneDrive to my W11 C: That suggested it was the path so I went back to OneDrive and up two levels so it was directly under OneDrive and now it works on both W10 and W11 when running from OneDrive.
<p>Note that Arduino IDE 1.8.19 AND 2.0.3 both fail with the same message. I'd bet it's a limitation in gcc library paths. 
<p>Try it - you'll LIKE it!
<p>BTW: I LOVE ArduinoJson! I use it to filter then parse api.weather.gov and it makes my "sketch" easy. Thank you**10**6

closed 2023-03-05 by bblanchon