bblanchon/ArduinoJson #1874: error: 'const void*' is not a pointer-to-object type

beliboba, 2023-02-19:
`error: 'const void*' is not a pointer-to-object type`
This error occurs when trying to compile sketch.
Im using arduino editor with arduino iot cloud.
Have a look at my code:
`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecure.h>
#include <ArduinoJson.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>
#include "thingProperties.h"

#define TFT_CS         12
#define TFT_RST        4
#define TFT_DC         5

#define stickVRX 16 // D0
#define stickVRY 0 // D3
#define stickSW 2 // D4

const char* ssid = "";
const char* password =  "";

const char* username = "";
const char* token = "";

const char* repo = "";

const char* host = "api.github.com";

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
HTTPClient http;
WiFiClient client;

template <typename T>
inline const T* pgm_read(const void* p) {
  return reinterpret_cast<const T*>(pgm_read_ptr(p));
}

void startup() {
  tft.fillScreen(ST77XX_BLACK);
  tft.setCursor(45, 120);
  tft.setTextSize(2);
  tft.println("Github Tracker");
}

void setup() {
  Serial.begin(115200);
  
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("connected");
  
  tft.init(240, 240, SPI_MODE3);
  tft.setRotation(2);
  delay(1500);
  initProperties();

  ArduinoCloud.begin(ArduinoIoTPreferredConnection);

  setDebugMessageLevel(2);
  ArduinoCloud.printDebugInfo();
  initProperties();
  ArduinoCloud.begin(ArduinoIoTPreferredConnection);
  setDebugMessageLevel(2);
  ArduinoCloud.printDebugInfo();
  
  startup();
  delay(500);
}

void notifications() {
  WiFiClientSecure client;
  if (!client.connect(host, 443)) {
    Serial.println("Connection failed");
    return;
  }
  if (client.verify(fingerprint, host)) {
    Serial.println("Certificate matches");
  } else {
    Serial.println("Certificate doesn't match");
    return;
  }
  client.print(String("GET /notifications?access_token=") + token + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: ESP8266\r\n" +
               "Connection: close\r\n\r\n");
  while (!client.available()) {
    delay(10);
  }
  String response = "";
  while (client.available()) {
    response += client.readString();
  }
  const size_t capacity = JSON_OBJECT_SIZE(10) + 1000;
  DynamicJsonDocument doc(capacity);
  DeserializationError error = deserializeJson(doc, response);
  if (error) {
    Serial.print("deserializeJson() failed: ");
    Serial.println(error.c_str());
    return;
  }
  JsonArray notifications = doc.as<JsonArray>();
  Serial.println("Latest notifications:");
  for (JsonObject notification : notifications) {
    Serial.println(notification["subject"]["title"].as<String>());
  }
}

void loop() {
  ArduinoCloud.update();
  // Get the latest notifications from GitHub every minute
  static unsigned long lastNotification = 0;
  if (millis() - lastNotification > 60000) {
    lastNotification = millis();
    notifications();
  }
}
`

bblanchon, 2023-02-20:
Related to https://github.com/arduino/ArduinoCore-API/pull/118.
Duplicate of #1442, #1519, #1536, #1591, #1676, and #1790.

If the error occurs in ArduinoJson (which I cannot tell because you didn't include the compiler output), you can work around this bug by defining [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`. BTW, the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/) would have told you that.

beliboba, 2023-02-23:
Thank you! Ill try it

closed 2023-03-05 by bblanchon