bblanchon/ArduinoJson #1875: Compilation error: no matching function for call to 'ArduinoJson6201_F1::JsonVariant::is<char*>()'

FARKIr, 2023-02-20:
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino: In function 'void pub(const char*, ArduinoJson6201_F1::JsonObject&)':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36: error: no matching function for call to 'ArduinoJson6201_F1::JsonVariant::is<char*>()'
     } else if (p.value().is<char*>()) {
                                    ^
In file included from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson.hpp:24,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:52:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: note: candidate: 'template<class T> typename ArduinoJson6201_F1::enable_if<ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value, bool>::type ArduinoJson6201_F1::VariantRefBase<TDerived>::is() const [with T = T; TDerived = ArduinoJson6201_F1::JsonVariant]'
       is() const {
       ^~
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: note:   template argument deduction/substitution failed:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6201_F1::enable_if<ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value, bool>::type ArduinoJson6201_F1::VariantRefBase<ArduinoJson6201_F1::JsonVariant>::is<T>() const [with T = char*]':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36:   required from here
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: error: no type named 'type' in 'struct ArduinoJson6201_F1::enable_if<false, bool>'
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: note: candidate: 'template<class T> typename ArduinoJson6201_F1::enable_if<(((! ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value) && (! ArduinoJson6201_F1::is_same<T, char*>::value)) && (! ArduinoJson6201_F1::is_same<T, char>::value)), bool>::type ArduinoJson6201_F1::VariantRefBase<TDerived>::is() const [with T = T; TDerived = ArduinoJson6201_F1::JsonVariant]'
   is() const {
   ^~
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: note:   template argument deduction/substitution failed:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6201_F1::enable_if<(((! ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value) && (! ArduinoJson6201_F1::is_same<T, char*>::value)) && (! ArduinoJson6201_F1::is_same<T, char>::value)), bool>::type ArduinoJson6201_F1::VariantRefBase<ArduinoJson6201_F1::JsonVariant>::is<T>() const [with T = char*]':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36:   required from here
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: error: no type named 'type' in 'struct ArduinoJson6201_F1::enable_if<false, bool>'

exit status 1

Compilation error: no matching function for call to 'ArduinoJson6201_F1::JsonVariant::is<char*>()'

FARKIr, 2023-02-20:
`C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino: In function 'void pub(const char*, ArduinoJson6201_F1::JsonObject&)':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36: error: no matching function for call to 'ArduinoJson6201_F1::JsonVariant::is<char*>()'
     } else if (p.value().is<char*>()) {
                                    ^
In file included from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson.hpp:24,
                 from c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:52:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: note: candidate: 'template<class T> typename ArduinoJson6201_F1::enable_if<ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value, bool>::type ArduinoJson6201_F1::VariantRefBase<TDerived>::is() const [with T = T; TDerived = ArduinoJson6201_F1::JsonVariant]'
       is() const {
       ^~
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: note:   template argument deduction/substitution failed:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6201_F1::enable_if<ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value, bool>::type ArduinoJson6201_F1::VariantRefBase<ArduinoJson6201_F1::JsonVariant>::is<T>() const [with T = char*]':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36:   required from here
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:93:7: error: no type named 'type' in 'struct ArduinoJson6201_F1::enable_if<false, bool>'
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: note: candidate: 'template<class T> typename ArduinoJson6201_F1::enable_if<(((! ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value) && (! ArduinoJson6201_F1::is_same<T, char*>::value)) && (! ArduinoJson6201_F1::is_same<T, char>::value)), bool>::type ArduinoJson6201_F1::VariantRefBase<TDerived>::is() const [with T = T; TDerived = ArduinoJson6201_F1::JsonVariant]'
   is() const {
   ^~
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: note:   template argument deduction/substitution failed:
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6201_F1::enable_if<(((! ArduinoJson6201_F1::ConverterNeedsWriteableRef<T>::value) && (! ArduinoJson6201_F1::is_same<T, char*>::value)) && (! ArduinoJson6201_F1::is_same<T, char>::value)), bool>::type ArduinoJson6201_F1::VariantRefBase<ArduinoJson6201_F1::JsonVariant>::is<T>() const [with T = char*]':
C:\Users\farka\.arduinoIDE\SKETCH\OMG\OMG.ino:389:36:   required from here
c:\Users\farka\.arduinoIDE\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:104:3: error: no type named 'type' in 'struct ArduinoJson6201_F1::enable_if<false, bool>'

exit status 1

Compilation error: no matching function for call to 'ArduinoJson6201_F1::JsonVariant::is<char*>()'`

bblanchon, 2023-02-21:
Hi @FARKIr,

This error happens when you write the following:

```c++
if (doc["key"].is<char*>())
```

Indeed, support for `char*` was deprecated in [6.18](https://arduinojson.org/news/2021/05/04/version-6-18-0/) and removed in [6.20](https://arduinojson.org/news/2022/12/26/arduinojson-6-20-0/).

To fix this error, you must replace `char*` with `const char*`, like so:

```c++
if (doc["key"].is<const char*>())
```

Best regards,
Benoit