bblanchon/ArduinoJson #1877: Make shrinkToFit() documents more usable.

LeedanJohnson, 2023-02-22:
1. Using a DynamicJsonDocument , I read in a document that may be very large.
2. I then call .shrinkToFit() immediately afterward to free up as much heap memory as possible.
3. After some processing, I try to replace the existing "updatedAt" field on my document with a new string of the same length.
4. doc.overflowed() now returns true, and serializing the document includes { "updatedAt": null }

Based on the documentation, I think I understand that this is due to the monotonic allocator never reusing memory, and shrinkToFit() releasing *all* unused memory.

Ideal solution (for my somewhat specific case):
    Allow replacing a value to reuse the same memory, provided the new value is smaller or equal to the existing value.
    This would probably be very involved, and may be impossible without unrealistic changes.

(Hopefully) More practical and general alternative:
    Allow shrinkToFit() to accept a uint argument specifying some amount of unused memory to retain.
    What if the document can't shrink at all, or can't shrink enough to offer the requested unused memory?
        I think it's reasonable that the retained memory be best-effort only.

LeedanJohnson, 2023-02-22:
In case anyone is interested, I am currently using this workaround.

`auto existingUpdatedAt = (char*) doc["updatedAt"].as<const char *>();
 auto newUpdatedAt = "2023-02-22T21:30:28.804Z";
 strncpy(existingUpdatedAt , newUpdatedAt , strlen(existingUpdatedAt ));`
 
 This is works because I know the strings will always be of equal length, but it certainly isn't good practice.

bblanchon, 2023-02-24:
Hi @LeedanJohnson,

Reusing the original value's memory is difficult because of the string deduplication. Indeed, if one alters the string, all nodes pointing to it are affected. There is no reference counter, and adding one would increase the footprint.

Adding a parameter to `shrinkToFit()` looks like an easy workaround.
Alternatively, I could add a `grow()` function that adds more space.

In any case, remember that ArduinoJson 6 was designed as a _serialization_ library; its job is to convert text to data and data to text.
The `JsonDocument` is supposed to be a short-lived vessel to contain this data; it never intended to replace `struct`, `std::vector`, and `std::map`.
I'll improve this in v7, but v6 will remain rigid in that regard.

Best regards,
Benoit

bblanchon, 2023-02-24:
Related to #1843

LeedanJohnson, 2023-02-24:
Thanks for getting back to me on this!

I figured the memory reuse idea is one that you'd have implemented long ago if it were trivial. Thanks for sharing some of the reasons why it's not.

My motivation behind advocating for adding a parameter to shrinkToFit() rather than adding a grow() function is due to limited heap memory. Unless I'm mistaken, all JsonDocuments rely on having a contiguous memory block. In that case, I would expect a grow() function to require allocating a new block of memory large enough to handle the new capacity requirement, copying the data over, then releasing the old memory. This is probably acceptable for most situations. In my case however, I'm working with a JsonDocument that could consume more than half of the available heap memory, which would make this fail. Specifying some amount of free memory to retain when calling shrinkToFit() should eliminate this issue because it requires no further memory allocation. (I think?)

I can absolutely understand if this kind of unique case (using more than half of the available heap memory for one Document) is so niche that you choose not to support it in favor of a more widely applicable solution. There are other (potentially 'better') ways I could handle this, including deserializing and re-serializing as you suggested.

Thanks for your suggestions and consideration.

bblanchon, 2023-07-29:
In branch [7.x](https://github.com/bblanchon/ArduinoJson/tree/7.x), `JsonDocument` has an elastic capacity, so you can still insert values after calling `shrinkToFit()`

closed 2023-07-29 by bblanchon