bblanchon/ArduinoJson #1878: Parsing JSON file without Test Mime

philippesikora, 2023-02-23:
I try to parse a JSON document from an HTTP response. The response is .cgx - type XML ' Content-Type TEXT / XML but format content is JSON as follows

{"TIME": "07:04" ,"date": "22/02/2023" , "SONDE2": 23.9 , "SONDE3": 22.6 , "SONDE4": 13.4 , "IINST1": 0 , "IINST2": 0 , "IINST3": 0 , "BASE_2": 002254963 , "BASE_3": 008599694 ,"SWITCH1": 0 , "SWITCH2": 1 , "SWITCH3": 0 , "SWITCH7": 0 , "SWITCH8": 0 , "SWITCH11": 0, "SWITCH12": 0, "SWITCH13": 0 , "ENTREE2": 0}

I got  DeserializationError, is it possible to use ArduinoJson library without Test Mime 

void ref_esp_cgx() 
{
  
  if (WiFi.status() == WL_CONNECTED) {
    http.useHTTP10(true);
    http.begin("http://192.168.8.207/espjson.cgx");     // get JSON file from WES device wih .xml extension
    http.setAuthorization("xxx", "xxx");

    http.GET();
    
    StaticJsonDocument<640>docref;                
    DeserializationError error = deserializeJson(docref, http.getStream());

    if (error) {                                       // Test if parsing succeeds 
      count_json_error++;                        
      return;
    }                                                  // Terminate the function
       
    ref_BASE_tic2 = docref["BASE_2"];  
    ref_BASE_tic3 = docref["BASE_3"];  
    TIME = docref["TIME"].as<String>();  
             
    http.end();
  }
}

philippesikora, 2023-02-24:
Dear Benoit,

I noticed my JSON file was invalid as an integer begins with a leading zero and change the data type from integer to string
The following JSON content from HTTP response is now valid 
I can see the file with JSON format from HTTP response with web browser Firefox (F12) - Network - Response
But unfortunately I still have "DeserializationError" issue with .cgx type in JSON format with ESP32

Input length: 315 from ArduinoJson Assistant
{"TIME": "07:04" ,"date": "22/02/2023" , "SONDE2": 23.9 , "SONDE3": 22.6 , "SONDE4": 13.4 , "IINST1": 0 , "IINST2": 0 , "IINST3": 0 , "BASE_2": "002254963" , "BASE_3": "008599694" ,"SWITCH1": 0 , "SWITCH2": 1 , "SWITCH3": 0 , "SWITCH7": 0 , "SWITCH8": 0 , "SWITCH11": 0, "SWITCH12": 0, "SWITCH13": 0 , "ENTREE2": 0}

 void ref_esp_cgx() {

    if (WiFi.status() == WL_CONNECTED) {
      http.useHTTP10(true);
      http.begin("http://192.168.8.207/espjson.cgx");    // get JSON file from WES device wih .cgx type in JSON format
      http.setAuthorization("xxx", "xxx");

      http.GET();

      StaticJsonDocument<344>docref;                
      DeserializationError error = deserializeJson(docref, http.getStream());

      if (error) {                                       // Test if parsing succeeds 
        count_json_error++;                        
        return;                                         //  Terminate the function
      }                                                  
   
      ref_BASE_tic2 = docref["BASE_2"].as<int>();
      ref_BASE_tic3 = docref["BASE_3"].as<int>();
      TIME = docref["TIME"].as<String>();

      http.end();
      }
}

Thank you for your support

bblanchon, 2023-02-24:
Hi @philippesikora,

What does the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/) say?

Best regards,
Benoit

philippesikora, 2023-02-24:
The value of DeserializationError is alternatively "EmptyInput" and "InvalidInput" at each http request iteration

bblanchon, 2023-02-25:
No big deal; you can share the two troubleshooter reports.

philippesikora, 2023-02-25:
Issue is fixed, JSON file was not valid. 
Compatible with .cgx type in JSON format working with ESP32.
Thank you

closed 2023-02-26 by bblanchon