bblanchon/ArduinoJson #1881: SerializedValue as JsonObject root

gonzabrusco, 2023-03-01:
Hello. I want to assing a SerializedValue directly to a JsonObject without adding a new key. 

For example:
I have a complex JSON generated with printf (because it has floating point numbers and I need to specify the amount of digits printed) and I want to make it the root of the JsonObject, can it be done?

```
char json[] = "{"timestamp":24245542424,"voltage_V_R":222.5,"current_A_R":445,"device/0/net_active_power_W_R":%244.55}"
StaticJsonDocument<512> jsonDoc;

JsonObject jsonObj = jsonDoc.to<JsonObject>();

jsonDoc = serialized(json);
```
That does not compile but changing the last line does:

`jsonDoc["hello"] = serialized(json);`

So my undestanding is that it's not possible to assign a serialized value to the root of the JsonObject (removing all the existents keys).

This is a simplified version of the code and I have no access to the JsonDocument that holds this JsonObject, so I need to be able to do this directly with the JsonObject.

Thanks!

bblanchon, 2023-03-02:
Hi @gonzabrusco,

The syntax you're looking for is:

```c++
doc.set(serialized(json));
```

Best regards,
Benoit

gonzabrusco, 2023-03-02:
Sorry, my mistake. I need to set a SerializedValue on a JsonObject (I don´t have access to the JsonDocument that owns it).

```
char json[] = "{"timestamp":24245542424,"voltage_V_R":222.5,"current_A_R":445,"device/0/net_active_power_W_R":%244.55}"
StaticJsonDocument<512> jsonDoc;

JsonObject jsonObj = jsonDoc.to<JsonObject>();

jsonObj = serialized(json); // <-- This is what I ment !
```

bblanchon, 2023-03-03:
I'm sorry, but it's only possible with a `JsonDocument` or a `JsonVariant`, not a `JsonObject`.

closed 2023-03-05 by gonzabrusco