bblanchon/ArduinoJson #1885: Wrappers over iterators

alikendarfen, 2023-03-05:
**Description**
My aim is to code wrappers around JsonObjectIterator and JsonArrayIterator, but it seems these classes are not available for users.
How can I do that?
Is there a structural reason to keeep these classes internal in ArduinoJson?

FYI, I need to wrap them in order to virtualize the messages interpretors (read/write virtual functions) so that the main code is independent of their implementation.

Thanks

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32
* Core/Framework: ESP32
* IDE: Arduino IDE 2.0.4

**Reproduction code**
```c++
  private:

    JsonObjectIterator iterator;

```

bblanchon, 2023-03-06:
Hi @alikendarfen,

Up to ArduinoJson 6.20, `JsonArrayIterator` and `JsonObjectIterator` were in the private namespace, but you could access them through `JsonArray::iterator` and `JsonObject::iterator`.
I don't see what prevents you from writing a wrapper class.

Best regards,
Benoit

alikendarfen, 2023-03-06:
Hello Benoit,

I did not know about JsonObject::iterator and though I can use JsonObjectIterator. So that's fine!

Thanks for your quick response.

Best regards

closed 2023-03-06 by alikendarfen