bblanchon/ArduinoJson #1887: DynamicJsonDocument `NoMemory` error with less than 30 bytes

MathewHDYT, 2023-03-08:
**Describe the bug**  
When using the `DynamicJsonDocument`, with a small amount of received bytes. From testing the edge case seems to occur at less than 30 bytes, the json can not be deserialized, it fails when trying to deserialize the inner the key-value pair with a `NoMemory` error.

I'm currently using the newest version of the library (6.20.1), but I did downgrade for testing purposes back to 6.19.3, it didn't change anything.

The received message length and message are completely correct, I'm simply printing them to the console as seen below in the output.

But as soon as I decrease the actual size of the value of the key-value pair so that it uses less than 30 bytes it will allocate the given amount + 1, but still fail with a `NoMemory` error.

As soon as I increase the capacity in the constructor to be at least 30 bytes the problem is fixed. Is there some kind of restriction on the `DynamicJsonDocument`, that requires that amount of memory to actually work, if yes adding that to the documentation would be nice.

Or is this some other underlying issue I didn't know about.

Because as soon as I replace this line on the below code snipped it work every time.

```c++
// Old not working below 30 bytes
DynamicJsonDocument jsonBuffer(length);

// Always works because we allocate at least 30 bytes
const size_t jsonSize = length < 30U ? 30U : length;
DynamicJsonDocument jsonBuffer(jsonSize);
```

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. `deserializeJson()` returns `NoMemory`
4. Increasing the capacity of the `JsonDocument` doesn't fix the issue
5. The `JsonDocument` is a `DynamicJsonDocument`
6. `doc.capacity()` returns the right value
7. Changing the Assistant settings doesn't solve the issue

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32
* Core/runtime: Arduino v2.0.6
* IDE: VSCode / PlattformIO

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
// Callback method called from PubSubClient onMessageCallback.
inline void process_attribute_request_message(char *topic, uint8_t *payload, const uint32_t length) {
    DynamicJsonDocument jsonBuffer(length);

     const DeserializationError error = deserializeJson(jsonBuffer, payload, length);
     if (error) {
         return;
     }
     JsonObjectConst data = jsonBuffer.template as<JsonObjectConst>();
     // Accesing json data....
}
```

**Program output**  

Expected output:

```
[TB] Requesting shared attributes transformed from (test) into json ({"clientKeys":"test"})
[TB] Callback onMQTTMessage from topic: (v1/devices/me/attributes/response/1)
Received payload: {"shared":{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}}
Sent length by MQTT Broker: 80
Allocated capacity: 80
Received payload: {"shared":{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}}
[TB] Received shared attribute request
[TB] Calling subscribed callback for response id (1)
hugeTestString
fdddddddddddfdddddddddddfdddddddddddfddddddddddd
{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}
[TB] Callback onMQTTMessage from topic: (v1/devices/me/attributes/response/2)
Received payload: {"client":{"test":118}}
Sent length by MQTT Broker: 23
Allocated capacity: 24
Received payload: {"client":{"test":118}}
[TB] Received shared attribute request
[TB] Calling subscribed callback for response id (2)
test
118
{"test":118}
```

Actual output:

```
[TB] Requesting shared attributes transformed from (test) into json ({"clientKeys":"test"})
[TB] Callback onMQTTMessage from topic: (v1/devices/me/attributes/response/1)
{"shared":{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}}
Sent length by MQTT Broker: 80
Allocated capacity: 80
{"shared":{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}}
[TB] Received shared attribute request
[TB] Calling subscribed callback for response id (1)
hugeTestString
fdddddddddddfdddddddddddfdddddddddddfddddddddddd
{"hugeTestString":"fdddddddddddfdddddddddddfdddddddddddfddddddddddd"}
[TB] Callback onMQTTMessage from topic: (v1/devices/me/attributes/response/2)
Received payload: {"client":{"test":118}}
Sent length by MQTT Broker: 23
Allocated capacity: 24
Received payload: {"client":{"test":118}}
Current deserialized text: {"client":{}}
[TB] Unable to de-serialize received json data with error (NoMemory)
```

bblanchon, 2023-03-09:
Hi @MathewHDYT,

How do you come up with `30`?
When I paste `{"shared":{"hugeTestString":"X"}}` into the Assistant, it tells me it needs 32 bytes.

Best regards,
Benoit

MathewHDYT, 2023-03-09:
@bblanchon That might be true, but what I meant with the 30 is not that the message is 30 bytes, but that it needs to be at least that size or it will produce the before mentioned `NoMemory` error.

But as soon as I always allocate at least 30 bytes for the `DynamicJsonDocument` it works from my testing.

bblanchon, 2023-03-10:
@MathewHDYT, you seem to think that the capacity of the `JsonDocument` is equal to the length of the input, but that's not the case. This capacity is used to store the internal representation of the JSON tree, which is usually very different from the serialized size. Please read the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/) for an explanation.

MathewHDYT, 2023-03-10:
Thanks for the information, would you have any idea what the easiest way to still guess the needed capacity is. Because I only know the size of payload. I do not know how many objects are contained in it.

Because I am receiving them from the `PubSubClient`.

 Is there some kind of library method that could help, or do I need to guess the capacity myself?

closed 2023-03-10 by MathewHDYT

bblanchon, 2023-03-11:
Please see [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)