bblanchon/ArduinoJson #1891: [5.x][bug] `createNestedObject(nullptr)` is `success()`full

MacDada, 2023-03-15:
**Describe the bug**  
`success()` after `createNestedObject(nullptr)` should be `false`, same as the behaviour of other `set()` methods.

**Environment**  
`PlatformIO`, `nodemcuv2` (esp8266) board, `Arduino` framework, `bblanchon/ArduinoJson @ ^5.13.4`

**Reproduction**  
```c++
void testArduinoJsonErrorHandling(AsyncWebServerRequest& request) {
    auto jsonBuffer = DynamicJsonBuffer{};
    auto& jsonRoot = jsonBuffer.createObject();

    char* uninitializedKey; // nullptr

    auto& jsonInnerObject = jsonRoot.createNestedObject(uninitializedKey);

    if (!jsonInnerObject.success()) {
        // I should see this log -> I don't -> I get an empty JSON object `{}` in the response (OK 200) instead
        logger.error("Failed to createNestedObject(uninitializedKey)");
    }

    if (!jsonInnerObject.set(uninitializedKey, "uninitializedKey value")) {
        // this is in the logs -> cool
        logger.error(PSTR("Failed to set uninitializedKey value"));
    }

    if (!jsonInnerObject.set("bar", "bar value")) {
        // this is not in the logs -> kinda OK, since this key-value pair is OK
        logger.error(PSTR("Failed to set bar value"));
    }

    auto* response = request.beginResponseStream("application/json");
    jsonRoot.printTo(*response);

    request.send(response);
}
```

**Possible solution**
Return `JsonObject::invalid()` instead of "normal" `JsonObject` when `set()` returns `false`: https://github.com/bblanchon/ArduinoJson/blob/5.x/src/ArduinoJson/JsonObjectImpl.hpp#L25

**Sidenotes**
I don't know if this is fixed in 6.x.

MacDada, 2023-03-15:
I can try to make a PR if it is decided this should be fixed.

MacDada, 2023-03-17:
> In C and C++, an uninitialized pointer is not set to null.
> Instead, it contains whatever data was present in RAM.

Good to know ;) Anyway, I was just trying to make a PoC of the (lack of) error handling, and this was enough :)

> I modified the function to return JsonObject::invalid() when the key is null

Cool, thanks!

> but the empty object is still in the buffer. I don't think I can fix this leak.

Hmm… But it is a short–living one, right? It is all deleted when `DynamicJsonBuffer` instance is gone?

bblanchon, 2023-03-17:
> Hmm… But it is a short–living one, right? It is all deleted when DynamicJsonBuffer instance is gone?

Exactly.

closed 2023-03-17 by bblanchon

MacDada, 2023-03-17:
Good enough for me – I don't intend to reuse the buffer.

Thanks for the fix!

I've now changed PIO spec to `https://github.com/bblanchon/ArduinoJson.git#5.x` and it is working :) Any chance to update [the PIO registry](https://registry.platformio.org/libraries/bblanchon/ArduinoJson/versions)? Interestingly, the latest version there is v.5.13.4, while the next one [contains quite a few changes](https://github.com/bblanchon/ArduinoJson/compare/v5.13.5..v5.13.4), not counting this fix.

commit 63c89f16 (2023-03-16): `JsonObject::createNestedObject()` returns `JsonObject::invalid()` if key is null (fixes #1891)