bblanchon/ArduinoJson #1897: Code no longer compiles on 6.21.0

mattncsu, 2023-03-22:
**Describe the bug**  
When I upgraded from 6.20.0 to, 6.21.0, my code no longer compiles. When I downgraded to 6.20.0, the error went away.  Here is the compiler errors:

```
In file included from c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:45:0,
                 from c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\Matt\Documents\Arduino\ESP32_Weatherflow\ESP32_Weatherflow\ESP32_Weatherflow.ino:3:
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp: In instantiation of 'ArduinoJson::V6210PB::DeserializationError::Code ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]':
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:36:23:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]'
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/deserialize.hpp:35:57:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::deserialize(ArduinoJson::V6210PB::JsonDocument&, TStream&&, Args ...) [with TDeserializer = ArduinoJson::V6210PB::detail::JsonDeserializer; TStream = unsigned char*&; Args = {unsigned int}]'
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:689:39:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::deserializeJson(ArduinoJson::V6210PB::JsonDocument&, TChar*, Args&& ...) [with TChar = unsigned char; Args = {unsigned int}]'
C:\Users\Matt\Documents\Arduino\ESP32_Weatherflow\ESP32_Weatherflow\ESP32_Weatherflow.ino:193:87:   required from here
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:74:9: error: request for member 'allowArray' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowArray())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:80:9: error: request for member 'allowObject' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowObject())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:87:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:93:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:98:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:108:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
```

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
I did not see the error in the troubleshooter

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32WROVER
* Core/runtime: Arduino-ESP32 2.0.7
* IDE: Arduino IDE 2.0.4

**Reproduction**  
The code listens for UDP broadcasts from a weather station and also requests the forecast from the API which returns a huge JSON payload.  These are the two functions that receive and parse the JSON.

```c++
void connectUDPReceiver()
{
  if (udp.listen(port))
  {
    udp.onPacket([](AsyncUDPPacket packet) {
      //StaticJsonDocument<384> doc;
      SpiRamJsonDocument doc(512);
      DeserializationError error = deserializeJson(doc, packet.data(), packet.length());

      if (error)
      {
        Serial.print(F("deserializeJson() failed: "));
        //Serial.println(error.f_str());
        return;
      }
      const char *type = doc["type"];

      if (strcmp(type, "obs_st") == 0)
      {

        JsonArray obs = doc["obs"][0];
        wx.obs_st_time = obs[0];
        wx.wind_lull = obs[1];
        wx.wind_avg = obs[2];
[... 220 lines cut ...]
      hourly[x].forecast_hourly_item_wind_direction_cardinal = forecast_hourly_item["wind_direction_cardinal"];
      hourly[x].forecast_hourly_item_wind_gust = forecast_hourly_item["wind_gust"]; // 5, 5, 5, 5, 5, 5, 5, 8, 8, 8, ...
      hourly[x].forecast_hourly_item_uv = forecast_hourly_item["uv"]; // 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 2, ...
      hourly[x].forecast_hourly_item_feels_like = forecast_hourly_item["feels_like"]; // 35, 34, 35, 33, 33, 33, 33, ...
      hourly[x].forecast_hourly_item_local_hour = forecast_hourly_item["local_hour"]; // 23, 0, 1, 2, 3, 4, 5, 6, 7, ...
      hourly[x].forecast_hourly_item_local_day = forecast_hourly_item["local_day"]; // 2, 3, 3, 3, 3, 3, 3, 3, 3, 3, ...
      x++;
    }
    Serial.println("Done with hourly forecasts");

Serial.println(hourly[1].forecast_hourly_item_conditions);  
    

  }
}

```

**Compiler output**    
If relevant, include the complete compiler output (i.e. not just the line that contains the error.)
See above

**Program output**  
If relevant, include the repro program output.
N/A - does not compile

mattncsu, 2023-03-22:
I'll try and create a MCVE tonight, thanks!

mattncsu, 2023-03-24:
Here's a (close to) MCVE that demonstrates the error.

```
#include <WiFi.h>
#include <AsyncUDP.h>
#include <ArduinoJson.h>
#include <HTTPClient.h>

const char * ssid = "xxxx";
const char * password = "xxxx";
const char * ntpServer = "192.168.1.100";

AsyncUDP udp;

const char* forecastURL = "https://swd.weatherflow.com/swd/rest/better_forecast?station_id=70&units_temp=f&units_wind=mph&units_pressure=inhg&units_precip=in&units_distance=mi&token=xxxxxx";

const char* root_ca= \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF\n" \
"-----END CERTIFICATE-----\n";

const int port = 50222;

struct SpiRamAllocator {
  void* allocate(size_t size) {
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
  }

  void deallocate(void* pointer) {
    heap_caps_free(pointer);
  }

  void* reallocate(void* ptr, size_t new_size) {
    return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);
  }
};

using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;

struct weather { //holds all the data received from the UDP broadcasts
  unsigned int rain_start; //rain start-time epoch seconds
} wx;

struct current_conditions{
//next block stores all the weather data received from the API
  long current_conditions_time; // 1677814324
} cc;

size_t len;
void connectUDPReceiver()
{
  if (udp.listen(port))
  {
    udp.onPacket([](AsyncUDPPacket packet) {
      SpiRamJsonDocument doc(512);
      DeserializationError error = deserializeJson(doc, packet.data(), packet.length());
      const char *type = doc["type"];
      if (strcmp(type, "rapid_wind") == 0)
      {
        JsonArray ob = doc["ob"];
      }
    });
  }
}

void getForecast(){
  if(WiFi.status()== WL_CONNECTED){
    HTTPClient http;
    http.useHTTP10(true);
    http.begin(forecastURL, root_ca);
[... 34 lines cut ...]
                 from C:\Users\Matt\AppData\Local\Temp\.arduinoIDE-unsaved2023223-28408-1sca3fk.e8vi\sketch_mar23a\sketch_mar23a.ino:3:
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp: In instantiation of 'ArduinoJson::V6210PB::DeserializationError::Code ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]':
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:36:23:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]'
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/deserialize.hpp:35:57:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::deserialize(ArduinoJson::V6210PB::JsonDocument&, TStream&&, Args ...) [with TDeserializer = ArduinoJson::V6210PB::detail::JsonDeserializer; TStream = unsigned char*&; Args = {unsigned int}]'
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:689:39:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::deserializeJson(ArduinoJson::V6210PB::JsonDocument&, TChar*, Args&& ...) [with TChar = unsigned char; Args = {unsigned int}]'
C:\Users\Matt\AppData\Local\Temp\.arduinoIDE-unsaved2023223-28408-1sca3fk.e8vi\sketch_mar23a\sketch_mar23a.ino:53:87:   required from here
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:74:9: error: request for member 'allowArray' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowArray())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:80:9: error: request for member 'allowObject' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowObject())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:87:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:93:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:98:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
c:\Users\Matt\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:108:9: error: request for member 'allowValue' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowValue())
         ^
```

bblanchon, 2023-03-24:
Thank you very much, @mattncsu.
This code compiles without any error with the following:

* Windows 11 Pro 22H2 on x64 processor
* Arduino IDE 2.0.4
* Board: ESP32 Wrover Kit (all versions)
* ESP32 core 2.0.7 (I tried 2.0.6 too)
* ArduinoJson 6.21.0 (I tested both git clone and install via library manager)
* No other library installed (I started from a fresh sketchbook)

I honestly don't know what's going on.
It seems we are executing the same code, yet, we get different results.

Can you try to uninstall and reinstall ArduinoJson 6.21.0?

bblanchon, 2023-03-24:
> I'll try the generic ESP32 WROVER definition tonight to see if that is the culprit

Thanks. Please let me know the outcome.

> I also have a million other libraries installed from years of playing with ESP8266, ESP32s, and ATMEGA/Tiny chips.

In the compiler output, you should see which libraries were included in the build.

mattncsu, 2023-03-25:
I confirmed the example compiles without error when I changed my board from "Inkplate 10" to "ESP32 Wrover Module"  I'm not sure what about the board definition it breaking it.  Here's the full compiler output if that helps.

```
Using board 'Inkplate10' from platform in folder: C:\Users\Matt\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.1
Using core 'Inkplate' from platform in folder: C:\Users\Matt\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.1
cmd /c if exist "C:\\Users\\Matt\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023223-28408-1sca3fk.e8vi\\sketch_mar23a\\partitions.csv" copy /y "C:\\Users\\Matt\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023223-28408-1sca3fk.e8vi\\sketch_mar23a\\partitions.csv" "C:\\Users\\Matt\\AppData\\Local\\Temp\\arduino\\sketches\\7091A5EAE8FD3817CECE8CDBB0C4B4FA\\partitions.csv"
cmd /c if not exist "C:\\Users\\Matt\\AppData\\Local\\Temp\\arduino\\sketches\\7091A5EAE8FD3817CECE8CDBB0C4B4FA\\partitions.csv" copy "C:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1\\tools\\partitions\\default.csv" "C:\\Users\\Matt\\AppData\\Local\\Temp\\arduino\\sketches\\7091A5EAE8FD3817CECE8CDBB0C4B4FA\\partitions.csv"
        1 file(s) copied.
Detecting libraries used...
"C:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\tools\\xtensa-esp32-elf-gcc\\1.22.0-96-g2852398-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/config" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/app_trace" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/app_update" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/asio" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/bootloader_support" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/bt" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/coap" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/console" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/driver" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/efuse" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp-tls" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp32" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_event" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_http_client" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_http_server" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_https_server" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/espcoredump" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/ethernet" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/expat" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/fatfs" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/freemodbus" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/freertos" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/heap" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/idf_test" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/jsmn" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/json" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardw [... cut ...] nsa-debug-module" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp-face" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp32-camera" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\1.0.1/tools/sdk/include/esp-face" "-IC:\\Users\\Matt\\AppData\\Local\\Arduino15\\packages\\Croduino_Boards\\hardware\\Inkplate\\

bblanchon, 2023-03-25:
It looks like you're using a deprecated core.

Form [Croduino-Board-Definitions-for-Arduino-IDE](https://github.com/SolderedElectronics/Croduino-Board-Definitions-for-Arduino-IDE):

> Croduino Board Definitions are not updated anymore. Please use new Soldered Dasduino Board Definitions for Dasduino and Inkplate boards. You can find them [here](https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE).

I tried with Soldered core, and your program compiles correctly.

Croduino uses GCC 5.2 (released in 2015), but ArduinoJson 6.21 require GCC 6 or above.
Soldered uses GCC 8.4 (released in 2020), the same used in the ESP32 core.

I excluded GCC 5 from ArduinoJson's test suite because it could not compile [Catch 2.x](https://github.com/catchorg/Catch2/tree/v2.x).
I could look for a workaround, but it's most likely a waste of time.
I think the best I can do is add an explicit compilation error when GCC is too old.

stif, 2023-03-26:
same with me. i am still using core 1.0.6 because of a ethernet network issue with core 2.0.4

platform = espressif32 @ ~3.5.0 (in platformio.ini)
```
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
```

it compiles fine with current core:

platform = espressif32 (in platformio.ini)
```
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

guess it is time to find the root cause of the ehternet network issue with Olimex PoE Hardware and Core 2.0.4 and stick with ArduinoJson version 6.20.1 until i solved it

ttlappalainen, 2023-03-26:
I have same problem. Currently I am stuck in Espressif 32 version 3.5.0.
```
  size_t len=packet.length();
  uint8_t *packetData=packet.data();
...
  deserializeJson(jsonBuffer,packetData,len);

```
Causes same error. I compiles fine on Espressif 32 version 6.1.0.

I think problem is that somehow compiler can not fit deserializeJson parameter list properly and thinks len is filter. I did not yet found how to force to select proper template.

mattncsu, 2023-03-27:
Switching to the Soldered core fixed it, thanks!

ttlappalainen, 2023-03-27:
Thanks a lot. Now compiles at least for me. I'll make functional tests later.

commit 9e9015f4 (2023-03-27): Fix compatibility with GCC 5.2 (fixes #1897)