bblanchon/ArduinoJson #1898: Compile Error with V6.21.0

stif, 2023-03-22:
**Description**
I have a Project which compiles fine with version 6.20.1 but fails with 6.21.0

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32
* IDE: PlatvormIO IDE v2.5.2 PlatformIO core 6.1.6

**Reproduction code**
```c++
constexpr size_t  jsonCapacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(3) + 36;

void WebService::onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void* arg, uint8_t *data, size_t len) {
  if (xSemaphoreTake(_smphr, 50) == pdTRUE) {
    if(type == WS_EVT_DATA) {
      Serial.print("Event Data received\n");
      AwsFrameInfo * info = reinterpret_cast<AwsFrameInfo*>(arg);
      if (info->final && info->index == 0 && info->len == len && info->opcode == WS_TEXT) {
        // the whole message is in a single frame and we got all of it's data
        StaticJsonDocument<jsonCapacity> doc;
        DeserializationError error = deserializeJson(doc, data, len); // <-line the compiler is complaining about

```

**Remarks**
Compile Error: In file included from .pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson.hpp:45:0,
                 from .pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson.h:9,
                 from lib/webservice/webservice.h:11,
                 from lib/webservice/webservice.cpp:1:
.pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp: In instantiation of 'ArduinoJson::V6210PB::DeserializationError::Code ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]':
.pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:36:23:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson::V6210PB::detail::VariantData&, TFilter, ArduinoJson::V6210PB::DeserializationOption::NestingLimit) [with TFilter = unsigned int; TReader = ArduinoJson::V6210PB::detail::Reader<unsigned char*, void>; TStringStorage = ArduinoJson::V6210PB::detail::StringMover]'
.pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:35:57:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::detail::deserialize(ArduinoJson::V6210PB::JsonDocument&, TStream&&, Args ...) [with TDeserializer = ArduinoJson::V6210PB::detail::JsonDeserializer; TStream = unsigned char*&; Args = {unsigned int}]'
.pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:689:39:   required from 'ArduinoJson::V6210PB::DeserializationError ArduinoJson::V6210PB::deserializeJson(ArduinoJson::V6210PB::JsonDocument&, TChar*, Args&& ...) [with TChar = unsigned char; Args = {unsigned int&}]'
lib/webservice/webservice.cpp:82:68:   required from here
.pio/libdeps/esp32-poe/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:74:9: error: request for member 'allowArray' in 'filter', which is of non-class type 'unsigned int'
         if (filter.allowArray())
         ^

stif, 2023-03-22:
Sorry i was redirected via arduinojson website and did not see the same issue already exists. this seems to be a duplicate from #1897 . will close this issue..

bblanchon, 2023-03-22:
Hi @stif,

Thank you very much for reporting this error.
Unfortunately, I don't manage to reproduce the error.
Can you provide an [MCVE](https://stackoverflow.com/help/mcve)?

Best regards,
Benoit