bblanchon/ArduinoJson #1899: API Response with multiple objects.  Unable to deserialize

cjlambiii, 2023-03-23:
I am sure it is something simple that I am missing but I haven't been able to figure it out yet.

`DynamicJsonDocument jsonDoc(4096);

void loop() {
  timeClient.update();
  
if (client.connect(server, 8000)) {
  client.print("GET /api/v1/ HTTP/1.1\r\n");
  client.print("Connection: close\r\n\r\n");

  while (!client.available()) {
    delay(100); 
  }

  String response = client.readString();
  Serial.println(client.read());  
  Serial.println(response);

  DeserializationError error = deserializeJson(jsonDoc, response);
  if (error) {
    Serial.print("Deserialization failed: ");
    Serial.println(error.c_str());
  } else {
    for (JsonVariant entry : jsonDoc["id"].as<JsonArray>()) {
      Serial.println(entry["id"].as<String>());
    }
  }
  client.stop();
  delay(30000);
}
}`

This is what I get.

`
11:11:32.825 -> -1
11:11:32.825 -> HTTP/1.1 200 OK
11:11:32.825 -> Date: Thu, 23 Mar 2023 18:11:31 GMT
11:11:32.825 -> Server: WSGIServer/0.2 CPython/3.10.5
11:11:32.825 -> Content-Type: application/json
11:11:32.825 -> Vary: Accept, Origin, Cookie
11:11:32.825 -> Allow: GET, POST, HEAD, OPTIONS
11:11:32.825 -> X-Frame-Options: DENY
11:11:32.825 -> Content-Length: 41
11:11:32.825 -> X-Content-Type-Options: nosniff
11:11:32.825 -> Referrer-Policy: same-origin
11:11:32.825 -> Cross-Origin-Opener-Policy: same-origin
11:11:32.825 -> 
11:11:32.825 -> [{"id":1,"author":1},{"id":2,"author":1}]
11:11:32.825 -> Deserialization failed: InvalidInput
`

bblanchon, 2023-03-24:
Hi @cjlambiii,

Please try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/).

Best regards,
Benoit

cjlambiii, 2023-03-25:
Thanks!

It turns out WiFiclient with wifinana streams the header as well as the
http payload.  After I stripped off the header it worked perfectly.

Cheers
-charles

bblanchon, 2023-03-25:
Awesome!
Thank you for the update, Charles.

closed 2023-03-25 by bblanchon