bblanchon/ArduinoJson #1900: JsonObject.set(newJsonObject) empties my object.

zekageri, 2023-03-30:
Hi. I want to copy a JsonObject into a nested object but after i use the `set()` method my object becomes empty.

**Here is my code:**

```cpp
boolean zSystem::getZoneObject(JsonObject& zone,const char* zoneID){
    accessLock();
    SpiRamJsonDocument zonesObj(ZONES_OBJ_SIZE);
    DeserializationError error = deserializeJson(zonesObj, (const char *)zones, ZONES_OBJ_SIZE);
    if( error ){
        Serial.printf(
            "[ZONES] - Failed to deserialize zone: %s. Error: %s\n",
            zoneID,
            error.c_str()
        );
        accessUnlock();
        return false;
    }
    accessUnlock();
    zone = zonesObj[zoneID];
    return !zone.isNull();
}

boolean zSystem::saveZoneObject(JsonObject &zoneObject, const char * zoneID){
    accessLock();
    boolean status = false;
    SpiRamJsonDocument zonesObj(ZONES_OBJ_SIZE);
    DeserializationError error = deserializeJson(zonesObj, (const char *)zones, ZONES_OBJ_SIZE);
    if( !error ){
        JsonObject zone = zonesObj[zoneID];
        if( zone.isNull() ){
            zone = zonesObj.createNestedObject(zoneID);
        }
        Serial.println();
        serializeJsonPretty(zonesObj,Serial);
        Serial.println();
        zone.set(zoneObject);
        if( serializeJson(zonesObj, zones, ZONES_OBJ_SIZE) ){
            startSave();
            status = true;
        }
        Serial.println();
        serializeJsonPretty(zonesObj,Serial);
        Serial.println();
    }
    accessUnlock();
    return status;
}

void WateringValve::saveToGlobalObject(JsonObject &config){
    JsonObject myZone;
    if( !zones.getZoneObject(myZone,zoneID) ){
        Serial.printf("[WATERING_VALVE] - Failed to get zone object\n");
        return;
    }
    JsonObject valves = myZone["wateringValves"];
    if( valves.isNull() ){ valves = myZone.createNestedObject("wateringValves"); }
    JsonObject valve = valves[id];
    if( valve.isNull() ){ valve = valves.createNestedObject(id); }
    if( valve.set(config) ){
        if( !zones.saveZoneObject(myZone,zoneID) ){
            Serial.printf("[WATERING_VALVE] - Failed to save zone object\n");
        }
    }
    //serializeJsonPretty(myZone,Serial);
}
```

I have a global zones object stored as a const char* in psram. This contains fresh informations of other objects inside each zone. If one component wants to save it's config, it calls saveToGlobalObject(). So the logic goes like this

- Component needs to save
- Component calls` zones.getZoneObject()` to get the zone that this particular component is belongs to.
- Component checks if the zones has the space for it, if not it creates it for itself.
- Component sets it's config into it's own space inside a zone
- Calls `zones.saveZoneObject()` to save the zone into the global zones
- Zones class will search the zone by zoneID and sets that zone.
- Zones class will start to save the serialized zones

`boolean zSystem::saveZoneObject(JsonObject &zoneObject, const char * zoneID)` prints to the serial the following

**Before set:**

```cpp
{
  "k_lfutlv84_7m207": {
    "name": "test",
    "isZone": true,
    "inGroup": false,
    "iconClass": "hsh-shader-teach",
    "iconColor": "#6495ed",
    "index": 1,
    "id": "k_lfutlv84_7m207"
  },
  "u_lfutlyde_sh8fw": {
    "name": "test2",
    "isZone": true,
    "inGroup": false,
    "iconClass": "hsh-shader-teach",
    "iconColor": "#6495ed",
    "index": 2,
    "id": "u_lfutlyde_sh8fw"
  }
}
```

**After set:**

```cpp
{
  "k_lfutlv84_7m207": {},
  "u_lfutlyde_sh8fw": {
    "name": "test2",
    "isZone": true,
    "inGroup": false,
    "iconClass": "hsh-shader-teach",
    "iconColor": "#6495ed",
    "index": 2,
    "id": "u_lfutlyde_sh8fw"
  }
}
```

As you can see the zone with id of **k_lfutlv84_7m207** become empty. It should have printed this after set

**Expected result**

```cpp
{
    "k_lfutlv84_7m207": {
        "name": "test",
        "isZone": true,
        "inGroup": false,
        "iconClass": "hsh-shader-teach",
        "iconColor": "#6495ed",
        "index": 1,
        "id": "k_lfutlv84_7m207",
        "wateringValves": {
            "b_lfutw6mu_k2a7o": {
                "id": "b_lfutw6mu_k2a7o",
                "name": "sdfg",
                "zoneID": "k_lfutlv84_7m207",
                "output": {
                    "pin": 4,
                    "address": 32
                },
                "createTime": 1680163376790,
                "modifyTime": 1680163376790
            }
        }
    },
    "u_lfutlyde_sh8fw": {
        "name": "test2",
        "isZone": true,
        "inGroup": false,
        "iconClass": "hsh-shader-teach",
        "iconColor": "#6495ed",
        "index": 2,
        "id": "u_lfutlyde_sh8fw"
    }
}
```

bblanchon, 2023-03-31:
Hi @zekageri,

Please provide an [MCVE](https://stackoverflow.com/help/mcve).

Best regards,
Benoit

zekageri, 2023-03-31:
Well. It works in the tests. I assume the problem is that I pass the object as a reference but I shouldn't. Sorry to bother you. I will try it in my project.

bblanchon, 2023-04-01:
Thank you for the link to the wokwi project, but a 149-line file is not really an MCVE.
This program contains a lot of code unrelated to this issue, and I don't have the time to read it.
Please shrink it down as much as possible if you want me to look at your issue.

zekageri, 2023-04-01:
Well. I think I figured it out. Thank you for the help! 149 line was necessary to demonstrate my use case. Since I pass objects between classes which are dinamically created and this could cause the issue. But it was really just an inattention.

closed 2023-04-01 by zekageri