bblanchon/ArduinoJson #1903: Use without Arduino

jputcu, 2023-04-03:
I'm currently using `ArduinoJson-6.18.5` without Arduino on AVR. I have provided the `FlashStringHelper` to support the flash string overloads.
Trying to upgrade to `ArduinoJson-6.21` I run into the issue that I require `Arduino.h` when `ARDUINOJSON_ENABLE_PROGMEM` is enabled.
The `pgm` defines are available on AVR.

Does the library need someting from Arduino to work with generic AVR flash/pgm data? Can we do without the `Arduino.h`?

bblanchon, 2023-04-04:
Hi @jputcu,

I need `Arduino.h` for the definition of `__FlashStringHelper`.
I cannot redefine it in ArduinoJson because [some cores define it in the `arduino` namespace](https://github.com/arduino/ArduinoCore-API/blob/844e4bf20c5fad8b89a2f307ba3416f61f512eaa/api/String.h#L44).

Why don't you add a dummy `Arduino.h`?

Best regards,
Benoit

bblanchon, 2023-04-05:
Hi @jputcu,

Please try the branch [issue1903](https://github.com/bblanchon/ArduinoJson/tree/issue1903).

Best regards,
Benoit

jputcu, 2023-04-05:
Hello

Thanks for the effort, will try it tomorrow.
BTW, I upgraded the `conan` receipe on `https://github.com/conan-io/conan-center-index/tree/master/recipes/arduinojson/conan-center-index` and saw you have an older repository for that. Perhaps it is best to remove it because it won't work with the newest `conan-2`.

Kind regards

Joris

bblanchon, 2023-04-06:
Hi Joris,

I archived the repo `bblanchon/conan-ArduinoJson`.
Do I need to do anything else?

Best regards,
Benoit

jputcu, 2023-04-06:
Hello Benoit

No, I think that's ok to avoid confusion.

Thanks
Joris

jputcu, 2023-04-06:
I'm trying to use the branch, but run into the issue on `ArduinoJson.hpp:17` due to `ARDUINOJSON_ENABLE_PROGMEM`.

bblanchon, 2023-04-07:
I added an extra check. Please try again.

jputcu, 2023-04-07:
It now compiles.

Just a few questions/remarks:
* I had to add `#define ARDUINOJSON_ENABLE_PROGMEM 1`, could it be autodetected on AVR?
* Coming from 6.18.5, my code size went from 72226 to 73174, checking the listing and map file for clues now.

bblanchon, 2023-04-07:
Finally someone who cares about code size!

If you look at the curves on [this page](https://arduinojson.org/news/2023/03/14/arduinojson-6-21-0/), you'll see that the ProgmemExample grew by about 200 bytes.
In your case, it's five times that, so there is probably a function that is inlined when it should not.

I must admit, PROGMEM support is not my top priority now that ESP32 and ARM are so popular.

How do you want me to detect we're on AVR, is `#ifdef __AVR__` suitable for you?

closed 2023-04-07 by bblanchon

jputcu, 2023-04-07:
I'm using `#ifdef __AVR`, but I'm unsure if all AVR have progmem. I've read somewhere in `ArduinoJson` that is detects the `pgm` defines, perhaps that is a better solution?

bblanchon, 2023-04-09:
The `pgm` macros are defined in `avr/pgmspace.h`, so there is a chicken-and-egg problem here.

I see that many projects are using `#ifdef __AVR___`; any reason why you use a different approach?

jputcu, 2023-04-11:
The documentation of `pgmspace.h` says it needs the `LPM` or `ELPM` instructions. So perhaps `__AVR__` and either one of the instructions?

bblanchon, 2023-04-12:
Fix available in [ArduinoJson 6.21.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.21.2)

commit 40daf56b (2023-04-07): Allow using PROGMEM outside of Arduino (fixes #1903)
commit 461cdaa8 (2023-04-21): Allow using PROGMEM outside of Arduino (fixes #1903)

Ported from 40daf56b5a1e4933c7852883c5f7902c3915f7d0