bblanchon/ArduinoJson #1907: Adding Key Value Pairs to Json Document v6 for Arduino ESP32 node

johnfischbeck, 2023-04-06:
I'm trying to add key value pairs to ArduinoJson v.6 document.  Please Help.

I've added the following below:
1.  Code Sample found to add values, but not key value pairs.
2. Output from Sample code with values added.
3. Output sample I need containing Key Value Pairs.

**Code Sample**

#include <ArduinoJson.h>

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial) continue;

  StaticJsonDocument<200> doc;

  doc["Timestamp"] = 1664313161;
  
  // Add an array.
  //
  JsonArray data = doc.createNestedArray("WiFiAccessPoints");
  data.add("This value");
  data.add(2.302038);

  // Generate the minified JSON and send it to the Serial port.
  //
  serializeJson(doc, Serial);
  // The above line prints:
  // {"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}

  // Start a new line
  Serial.println();

  // Generate the prettified JSON and send it to the Serial port.
  //
  serializeJsonPretty(doc, Serial);
  // The above line prints:
  // {
  //   "sensor": "gps",
  //   "time": 1351824120,
  //   "data": [
  //     48.756080,
  //     2.302038
  //   ]
  // }
}

void loop() {
  // not used in this example
}

**Sample Program output from above**  
{
  "Timestamp": 1664313161,
  "WiFiAccessPoints": [
    "This value",
    2.302038
  ]
}

****Desired output containing key value pairs****  
{
  "Timestamp":1664313161,
    "WiFiAccessPoints":[
        **{
        "MacAddress":"A0:EC:F9:1E:32:C1",
        "Rss":-75,
        "MacAddress":"A0:EC:F9:1E:32:C1",
        "Rss":-75,
        "MacAddress":"A0:EC:F9:1E:32:C1",
        "Rss":-75
        }**
      ]
}

BlueAndi, 2023-04-07:
https://arduinojson.org/v6/api/jsonarray/createnestedobject/

johnfischbeck, 2023-04-07:
Thanks @BlueAndi , One tab over I found the holy grail of json construction parsing in the ArduinoJson Assistant.  I have a hard time getting my head around json formatting so this tool is a godsend for me.

https://arduinojson.org/v6/assistant/#/step1

closed 2023-04-07 by johnfischbeck