bblanchon/ArduinoJson #1908: ArduinoJson wont compile on ESP-IDF

Aggeloz, 2023-04-07:
**Describe the issue**  
After including the library and trying to build/compile my code i get an error saying that the library is in cpp but my compiler is not

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP32S3]
* Core/runtime: [e.g. ESP32S2 core for ESP IDF]
* IDE: [e.g. ESP IDF]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c
#include <ArduinoJson.h>
```

**Program output**  
If relevant, include the program output.

Actual output:

```
error: #error ArduinoJson requires a C++ compiler, please change file extension to .cc or .cpp
   15 | #error ArduinoJson requires a C++ compiler, please change file extension to .cc or .cpp
```

bblanchon, 2023-04-08:
Hi @Aggeloz,

Did you try to change the extension of your source file from `.c` to `.cpp`?

Best regards,
Benoit

Aggeloz, 2023-04-08:
I thought i did but i seem to have messed something up in vscode and saved both one file with .c and one with .cpp. Thanks!

closed 2023-04-08 by Aggeloz