bblanchon/ArduinoJson #1909: Merging 2 JsonVariants fails

vortex314, 2023-04-12:
Hi Benoit,

this worked some time ago. Now it gives a compile error.

```
void merge(JsonVariant dst, JsonVariantConst src) {
  if (src.is<JsonObjectConst>()) {
    for (JsonPairConst kvp : src.as<JsonObjectConst>()) {
      if (dst[kvp.key()])
        merge(dst[kvp.key()], kvp.value());
      else
        dst[kvp.key()] = kvp.value();
    }
  } else {
    dst.set(src);
  }
}
```
gives 
```
/home/lieven/workspace/limero/linux/ConfigFile.cpp:92:30: error: no matching function for call to ‘ArduinoJson6172_71::VariantRef::is<ArduinoJson::JsonObjectConst>()’
   92 |   if (src.is<JsonObjectConst>()) {
```
Changing JsonVariantConst to JsonVariant didn't help either.

See also https://github.com/bblanchon/ArduinoJson/issues/1783

bblanchon, 2023-04-14:
Hi @vortex314,

Indeed [`JsonVariant::is<JsonObjectConst>()`](https://arduinojson.org/v6/api/jsonvariantconst/is/) was added in version 6.18.0 (see #1412).
You're currently using 6.17.2.

Best regards,
Benoit

vortex314, 2023-04-14:
Hi Benoit,

thanks for your swift response. Nice that you can see the version of ArduinoJson used in the compiler error message !
I'll try to get the latest version and check this.

Lieven

closed 2023-04-14 by vortex314