bblanchon/ArduinoJson #1914: Interference between ArduinoJson 6.21xx and Blynk 1.2.0

3KUdelta, 2023-04-17:
Hi, I am using ArduinoJson 6.21.xx in combination with Blynk 1.2.0 (//https://github.com/blynkkk/blynk-library) and I made the following interesting observation. As soon as I go to 6.21.xx I get two main compilation errors (details below). One on blynk side

`/Users/mst/Documents/Arduino/libraries/Blynk/src/Blynk/BlynkHandlers.h:25:13: error: expected unqualified-id before numeric constant
/Users/mst/Documents/Arduino/libraries/Blynk/src/Blynk/BlynkHandlers.h:25:13: error: expected '{' before numeric constant
   25 | #define V6  6
      |             ^`

and one on ArduinoJson side:

`Solar-WiFi-Weather-Station-V2.4:377:3: error: 'serializeJson' was not declared in this scope
  377 |   serializeJson(jsonDoc, jsonBuffer);
      |   ^~~~~~~~~~~~~
exit status 1
'StaticJsonDocument' was not declared in this scope`

Just to be clear: If I switch back to Versions 6.20.x and earlier it is working ok with no compilation errors. I tested also on the Blynk side - as soon as I use `#include <BlynkSimpleEsp8266.h>` together with any ArduinoJson example the error occurs.

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266
* Core/runtime: ESP8266 core for Arduino v3.0.2
* IDE: Arduino IDE 1.8.19

**Reproducible code example causing the error**
I am using your JsonParserExample. Another interesting observation: if I change the sequence of the two libraries, the error disappears. 

// ArduinoJson - https://arduinojson.org
// Copyright © 2014-2023, Benoit BLANCHON
// MIT License
//
// This example shows how to deserialize a JSON document with ArduinoJson.
//
// https://arduinojson.org/v6/example/parser/

#include <BlynkSimpleEsp8266.h>
#include <ArduinoJson.h>

void setup() {

  Serial.begin(9600);
  while (!Serial) continue;
  
  StaticJsonDocument<200> doc;

  char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  DeserializationError error = deserializeJson(doc, json);
  
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }
  const char* sensor = doc["sensor"];
  long time = doc["time"];
  double latitude = doc["data"][0];
  double longitude = doc["data"][1];

  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {
  // not used in this example
}

**Compiler output**    
Arduino: 1.8.19 (Mac OS X), Board: "LOLIN(WEMOS) D1 mini Pro, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 16MB (FS:14MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"
[... 608 lines cut ...]
/Users/mst/Documents/Arduino/libraries/Blynk/src/Blynk/BlynkHandlers.h:25:13: error: expected unqualified-id before numeric constant
/Users/mst/Documents/Arduino/libraries/Blynk/src/Blynk/BlynkHandlers.h:25:13: error: expected '{' before numeric constant
   25 | #define V6  6
      |             ^
/Users/mst/Documents/Arduino/libraries/Blynk/src/Blynk/BlynkHandlers.h:25:13: error: expected unqualified-id before numeric constant
/Users/mst/Documents/Arduino/sketches/Solar_WIFI_Weather-Station/Solar-WiFi-Weather-Station-V2.4/Solar-WiFi-Weather-Station-V2.4.ino: In function 'void setup()':
Solar-WiFi-Weather-Station-V2.4:353:3: error: 'StaticJsonDocument' was not declared in this scope
  353 |   StaticJsonDocument<512> jsonDoc;
      |   ^~~~~~~~~~~~~~~~~~
Solar-WiFi-Weather-Station-V2.4:353:27: error: 'jsonDoc' was not declared in this scope
  353 |   StaticJsonDocument<512> jsonDoc;
      |                           ^~~~~~~
Solar-WiFi-Weather-Station-V2.4:354:3: error: 'JsonObject' was not declared in this scope
  354 |   JsonObject rootObj = jsonDoc.createNestedObject("root");
      |   ^~~~~~~~~~
Solar-WiFi-Weather-Station-V2.4:358:3: error: 'rootObj' was not declared in this scope
  358 |   rootObj["temperature"] = adjusted_temp;
      |   ^~~~~~~
Solar-WiFi-Weather-Station-V2.4:377:3: error: 'serializeJson' was not declared in this scope
  377 |   serializeJson(jsonDoc, jsonBuffer);
      |   ^~~~~~~~~~~~~
exit status 1
'StaticJsonDocument' was not declared in this scope

Hope this helps. Best, Marc

bblanchon, 2023-04-18:
Hi @3KUdelta,

Thank you very much for reporting this issue.
`#define V6 6` was indeed interfering with the macro that generates ArduinoJson's namespace.

Best regards,
Benoit

closed 2023-04-18 by bblanchon

bblanchon, 2023-07-23:
Fix available in [ArduinoJson 6.21.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.21.3).

commit 52d8a65c (2023-04-18): Fix compatibility with the Blynk libary (fixes #1914)
commit 95f5d9d1 (2023-04-21): Fix compatibility with the Blynk libary (fixes #1914)

Ported from 52d8a65cbccd5b3e873bf13fffbc66650c7d96d2