bblanchon/ArduinoJson #1919: Google GeocodeAPI Parsing

TheRoyalGuard001, 2023-04-26:
I understand that I have to have the troubleshooter here but I can't upgrade from V5->V6 because of some complicated reasons that I don't understand. I hope that I can still get help without it.
Im completely new to this kind of stuff and this is my first large project (9th grade) so please give me some slack if I don't know what things mean or I use words incorrectly.

**Describe the issue**  
Although I went through the steps on the website about how to fix a parsing error, root.success() will always return false and I cannot read the return document. I am using google reverse geocode API and have attached the return file below (copied and pasted the url into a web browser).

**Troubleshooter report**  
N/A

**Environment**  
Here is the environment that I'm using':
ESP8266
* Core/runtime: Unknown
* IDE: Arduino IDE 1.8.19

**Reproduction**  
while (client.available()) {
    String line = client.readStringUntil('\r');
    JsonArray& root = jsonBuffer.parse(line);
    if (root.success()) {
      const char*  formatted_address = root[1]["results"]["formatted_address"];
      Serial.print("formatted_address = ");
      Serial.println(formatted_address);...
    } else {
      Serial.println ("root !success");
    }

**Program output**  

Expected output:

```
//address
```

Actual output:

```
root !success
```

Thanks for the help and dealing with my naivety.

[image]

Full code:
`
#include <ESP8266HTTPClient.h>
#include "ESP8266WiFi.h"
WiFiServer server(80); //default server port is 80
char myssid[] = "iPhone (2)";         // your network SSID (name)
char mypass[] = "ESD1223334444";          // your network password
#define ARDUINOJSON_DECODE_UNICODE 1
#include <ArduinoJson.h>
String key = "AIzaSyDbMnxUsvv7xa-9_ewd_aOrEaVddKg6xu4";
const char* Host = "www.googleapis.com";
String thisPage = "/maps/api/geocode/json?latlng=";

 void getGeocode(double lat, double lng) {
    DynamicJsonBuffer jsonBuffer;
  Serial.println("Geocode start");
  String geocodeJson;
  geocodeJson += "https://maps.googleapis.com/maps/api/geocode/json?latlng=";
  geocodeJson += String(lat,6) + "," + String(lng,6);
  geocodeJson += "&key=" + key;

  
  Serial.println("");
  WiFiClientSecure client;
  client.setInsecure();  
  
  Serial.print("Requesting URL: ");
  Serial.println(geocodeJson);
  Serial.println(" ");
  if (client.connect(Host, 443)) {
    Serial.println("Connected");
    client.println("POST " + thisPage + key + " HTTP/1.1");
    client.println("Host: " + (String)Host);
    client.println("Connection: close");
    client.println("Content-Type: application/json");
    client.println("User-Agent: Arduino/1.0");
    client.print("Content-Length: ");
    client.println(geocodeJson.length());
    client.println();
    client.print(geocodeJson);
    delay(500);
  } else {
      Serial.println("Cannot Connect");
  }

  while (client.available()) {
    String line = client.readStringUntil('\r');
    JsonArray& root = jsonBuffer.parse(line);
    if (root.success()) {
      const char*  formatted_address = root[1]["results"]["formatted_address"];
      const char*  address_street_number = root[0]["results"]["address_components"][0]["long_name"];
      const char*  address_street = root[0]["results"]["address_components"][1]["short_name"];
      const char*  address_neighborhood = root[0]["results"]["address_components"][2]["long_name"];
      const char*  address_sublocality = root[0]["results"]["address_components"][3]["long_name"];
      const char*  address_administrative_area_level_2 = root[0]["results"]["address_components"][4]["long_name"];
      const char*  address_administrative_area_level_1 = root[0]["results"]["address_components"][5]["long_name"];
      const char*  address_country = root[0]["results"]["address_components"][6]["long_name"];
      const char*  address_postal_code = root[0]["results"]["address_components"][7]["long_name"]; 
      Serial.print("formatted_address = ");
      Serial.println(formatted_address);     
      Serial.print("street_number = ");
      Serial.println(address_street_number);    
      Serial.print("street = ");
      Serial.println(address_street);    
      Serial.print("neighborhood = ");
      Serial.println(address_neighborhood);    
      Serial.print("sublocality = ");
      Serial.println(address_sublocality);    
      Serial.print("administrative_area_level_2 = ");
      Serial.println(address_administrative_area_level_2);    
      Serial.print("administrative_area_level_1 = ");
      Serial.println(address_administrative_area_level_1);    
      Serial.print("country = ");
      Serial.println(address_country);    
      Serial.print("postal_code = ");
      Serial.println(address_postal_code);          
    } else {
      Serial.println ("root !success");
    }
  }

  DeserializationError error = deserializeJson(doc, input);

Serial.print("deserializeJson() returned ");
Serial.println(error.c_str());

  Serial.println("closing connection");
  Serial.println();
  client.stop();

}

void setup(){
    Serial.begin(9600);
  //********************WIFI SETUP AND CONNECTION******************
  Serial.println("Start");
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  Serial.println("Setup done");
  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(myssid);
  WiFi.begin(myssid, mypass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(".");
  server.begin();

}

void loop(){
  getGeocode(40.714224,-73.961452);
  delay (20000);
}

`

bblanchon, 2023-04-27:
Hi @TheRoyalGuard001,

> This error occurs when you index a [JsonObject](https://arduinojson.org/v5/api/jsonobject/) with an integer instead of a string.

Please try:

```diff
- const char* formatted_address = root[1]["results"]["formatted_address"];
+ const char* formatted_address = root["results"]["formatted_address"];
```

Best regards,
Benoit

TheRoyalGuard001, 2023-04-27:
Thanks for the response!
This did fix my problem of " if (Internals::StringTraits::equals(key, it->key)) break;" but it is still printing "root !success". Any ideas?

(I convinced my teacher to buy your textbook and am making my way through it now. Thanks for creating great documentation, its really helpful)

TheRoyalGuard001, 2023-04-27:
Did not mean to close it.

bblanchon, 2023-04-29:
Please read [Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/)

TheRoyalGuard001, 2023-04-30:
1:
a) has no headers
b)Using Http 1.1
c)Timeout does not fix issue
2-4: Used ArduinoJson assistant so shouldn't be problem
5: Json is object
6: Nesting limit is not too high

Could the problem be in the headers I send? Ive touched up the code a little bit and have included that below.
[This](https://maps.googleapis.com/maps/api/geocode/json?latlng=40.881770,-74.004930&key=AIzaSyDbMnxUsvv7xa-9_ewd_aOrEaVddKg6xu4) is the page that pops up when pasting geocodingJson into my web browser.

`#include <ESP8266HTTPClient.h>
#include "ESP8266WiFi.h"
WiFiServer server(80); //default server port is 80
char myssid[] = "SSID";         // your network SSID (name)
char mypass[] = "PASS";          // your network password

#include <ArduinoJson.h>
String key = "KEY";
const char* Host = "www.googleapis.com";
String thisPage = "/maps/api/geocode/json?latlng=";
const char* formatted_address;
int status = WL_IDLE_STATUS;

 void getGeocode(double lat, double lng) {
  const int BUFFER_SIZE = 12288;
  DynamicJsonBuffer jsonBuffer (BUFFER_SIZE);
  
  Serial.println("Geocode start");
  String geocodeJson;
  geocodeJson += "https://maps.googleapis.com/maps/api/geocode/json?latlng=";
  geocodeJson += String(lat,6) + "," + String(lng,6);
  geocodeJson += "&key=" + key;

  
  Serial.println("");
  WiFiClientSecure client;
  client.setInsecure();  
  client.setTimeout(10000);
  
  Serial.print("Requesting URL: ");
  Serial.println(geocodeJson);
  Serial.println(" ");
  if (client.connect(Host, 443)) {
    Serial.println("Connected to Host");
    client.println("POST " + thisPage + key + " HTTP/1.1");
    client.println("Host: " + (String)Host);
    client.println("Connection: close");
    client.println("Content-Type: application/json");
    client.println("User-Agent: Arduino/1.0");
    client.print("Content-Length: ");
    client.println(geocodeJson.length());
    client.println();
    client.print(geocodeJson);
    delay(500);
  } else {
      Serial.println("Cannot Connect");
  }

  while (client.available()) {
    auto Line = client.readStringUntil('/n');
    JsonObject& root = jsonBuffer.parseObject(Line);
    
    if (root.success()) {
      auto formatted_address = root["results"]["formatted_address"];
      break;
    } else {
      Serial.println ("root !success");
      break;
    }
  }

      Serial.print("formatted_address = ");
      Serial.println(formatted_address);       
  Serial.println("closing connection");
  Serial.println();
  client.stop();

}

void setup(){
    Serial.begin(9600);
  //********************WIFI SETUP AND CONNECTION******************
  Serial.println("Start");
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  Serial.println("Setup done");
  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(myssid);
  WiFi.begin(myssid, mypass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(".");
  server.begin();

}

void loop(){
  getGeocode(40.714224,-73.961452);
  delay (20000);
}`

bblanchon, 2023-05-03:
> a) has no headers

I don't think your code skips the headers correctly.
Did you try to print `Line`?
Why don't you replicate [JsonHttpClient.ino](https://arduinojson.org/v5/example/http-client/)?

closed 2023-06-16 by bblanchon