bblanchon/ArduinoJson #1922: error: no matching function for call to 'convertToJson(const char&, ArduinoJson::V6212JB::JsonVariant&)'

p-hu, 2023-05-05:
**Description**
Hi, I feel rather stupid to report such a profound issue. I followed the instructions which are pretty easy and it did not help.
This is the function I added:
```
inline void convertToJson(const char& src, JsonVariant dst) {
  char buf[32];
  snprintf(buf, sizeof(buf), "%c", src);
  dst.set(buf);
}
```
The strange thing is, the code used to work just perfectly until I involuntary updated the PlatformIO configuration. In that moment, quite a lot of packages got updated and I can't figure out, which ones.
I mean, wouldn't it be kind of to be expected that ArduinoJson is perfectly capable of serialize a simple char?
I can go around the issue by putting my char into a const char* but it just feels wrong.
Any hint of what the real reason behind this issue could be?

**Troubleshooter's report**
1. The issue happens at compile time
2. Error says "no matching function for call to ..."
3. Error says "no matching function for call to `convertToJson(...)`"
4. Converting the value doesn't fix the issue

**Environment**
* Microcontroller: ESP32S2
* Core/Framework: Arduino / espressif32 (#v4.2.0 - #v6.2.0) / board = adafruit_qtpy_esp32s2
* IDE: VS PlatformIO Core 6.1.6·Home 3.4.3

**Reproduction code**
```c++
class Settings
{
public:
char get_char(ESettingName id) const {return getReference(id).as_char();}
...
};

class WebServer
{
public:
  template <size_t N>
  void append_flexEVSEsettings(StaticJsonDocument<N>& doc)
  {
    StaticJsonDocument<N>& doc
    auto doc_AP = doc["AP"];
    doc_AP["_ApTimeOutMode"] = settings_.get_char(Settings::ESettingName::_ApTimeOutMode);
  }
...
};

WebServer webServer_;
StaticJsonDocument<3000> response_doc;

webServer_.append_flexEVSEsettings(response_doc);
```

**Remarks**
I made sure that my extra convertToJson was actually compiled by just adding a syntax error in front of the function declaration and got the expected error from the compiler.
I can not figure out why the compiler would not be satisfied with my convertToJson function. But much more curious actually is, why do I get the error in the first place? I'm sure ArduinoJson would be able to encode a simple char. Other types work perfectly, I mean: bool, int, unsigned int, const char*, doubles, String etc.

Thanks a lot for your time to read this!
BR
Phil

bblanchon, 2023-05-06:
Hi Phil,

Since `char` is a built-in type, [ADL](https://en.cppreference.com/w/cpp/language/adl) ignores it and searches only in `JsonVariant`'s namespace, `ArduinoJson`.
Here is an updated (and optimized) version of your converter:

```c++
namespace ArduinoJson {
void convertToJson(char c, JsonVariant var) {
  char buf[] = {c, 0};
  var.set(buf);
}
}
```

Please note that older versions of ArduinoJson treated `char`s as integers, which is not what we do in this converter.
Support for `char` was deprecated in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/#support-for-naked-char-removed) then removed in [ArduinoJson 6.20.0](https://arduinojson.org/news/2022/12/26/arduinojson-6-20-0/#removed-support-for-char-and-char).

Best regards,
Benoit

p-hu, 2023-05-06:
Thanks a lot! It works, yeay :-)

bblanchon, 2023-05-07:
You're welcome, Phil!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2023-05-07 by bblanchon