bblanchon/ArduinoJson #1925: How to programmatically add key values

cyberbilly, 2023-05-26:
The following function doesn't work, from my experimentation It appears the the key is not set,  For example if I call 
sendMessage("warning", "State", "unknown");  I would like to get {"Purpose":"warning":"state":"unknown}.
What I get is {}.

void sendMessage( String level, String key, String value){
	DynamicJsonDocument msg(64);
	msg["Purpose"] = level;
	msg[key] = value;
	serializeJson(msg,  Serial);
    Serial.println();		
}

bblanchon, 2023-05-28:
Hi @cyberbilly,

When you use `String` for keys, the `JsonDocument` copies the string, consuming more space in the memory pool.
To fix this issue, you need to increase the capacity of the memory pool (to 96 in this case).

By default, the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) assumes you use only string literals for keys, so it doesn't count the extra bytes needed to copy them. You can change this in the "Tweaks" section of Step 3.

[image]

Best regards,
Benoit

cyberbilly, 2023-05-29:
Thank you for your assistance.   You said " By default, the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) assumes you use only string literals for keys"  Which I am doing everywhere,  Except for this one little generalized send message function.  In this case at coding time i don't know the key values I want to messages so I would like to use arguments to the function.   I haven't been able to accomplish this.  doc[key] = value.    doesn't work.

cyberbilly, 2023-05-29:
Okay the print on your screen capture was not readable - who decided light gray was a good font color? and the print is to small </excuses >   Any suggestions for getting to this setting on a command line only set up?

closed 2023-06-16 by bblanchon