bblanchon/ArduinoJson #1926: Call Function with DynamicJsonDocument Reference as Argument

MTBfs, 2023-06-04:
Hello,
I've the following problem when using ArduinoJson.

**Description**
I want to pass an reference of an DynamicJasonDocument to an function.
Everything seems to be ok, but the linker throws the following error:
undefined reference to `sdCard::listAudioLib(ArduinoJson::V6212PB::BasicJsonDocument<ArduinoJson::V6212PB::DefaultAllocator>&)'

If I remove the reference from the needed arguments, everything proceeds fine.

It takes also no difference if I use an pointer instead of an reference.

ArduinoJson is working correctely when I do everything in the same function, but this is not possible for this class and function.

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32-LyraT V4.3
* IDE: ESP-IDF v4.4.4

**Reproduction code**
```c++
class sdCard{
   public:
      bool listAudioLib(DynamicJsonDocument &libJson);
}

bool sdCard::listAudioLib(DynamicJsonDocument &libJson){
   return true;
}

extern "C" void app_main(void){
   sdCard *mSD = new sdCard();
   DynamicJsonDocument test(2048);
   mSd->listAudioLib(test);
}

```

**Remarks**
I'm using C17 and C++17, int VS Code with ESP-IDF Extension.

Thank you very much for helping me.
I hope there is a way to do this.

bblanchon, 2023-06-05:
Hi @MTBfs,

Your reproduction code compiles fine; see: https://wandbox.org/permlink/krD1i7apWfLWF7GQ

The linker error you see usually means that you have different definitions in multiple compilation units.
Ensure all the definitions of `listAudioLib()` are identical (except if you want to overload, of course).

Best regards,
Benoit

MTBfs, 2023-06-06:
Hello Benoit,

thank you for your help and answer.
Ok, that's somewhat weird, because I only copied the parts as they are from my whole code.
There are only two things I changed, I copied all parts from different files in one and I removed unnessecary code.

So I will build up from an blank project and will test that.
After that I'll come back to you, hopefully with no more errors.

Thank you very much.

Regards,
Fabian

MTBfs, 2023-06-06:
Hello Benoit,

I tried many hours, how to get rid of this error.
I'm using for other parts of my code the arduino core as component, for that I've required arduino in the CMake configuration (\main\CMakeLists.txt). If I remove the arduino requirement I can compile without any errors.

### Here are all contents of my project folder:

**\CMakeLists.txt**
```
cmake_minimum_required(VERSION 3.5)

include($ENV{ADF_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(sd_card)

set(CMAKE_C_STANDARD 17)
set(CMAKE_CXX_STANDARD 17)
```

**\main\CMakeLists.txt - _Not Working_**
```
idf_component_register(SRCS "main.cpp"
                    INCLUDE_DIRS "."
                    REQUIRES ArduinoJson arduino sdCard)
```

**\main\CMakeLists.txt - _Working, but not possible in my project_**
```
idf_component_register(SRCS "main.cpp"
                    INCLUDE_DIRS "."
                    REQUIRES ArduinoJson sdCard)
```

**\main\main.cpp**
```
#include "ArduinoJson.h"
#include "sdCard.hpp"

extern "C" void app_main(void){
   sdCard *mSd = new sdCard();
   DynamicJsonDocument test(2048);
   mSd->listAudioLib(test);
}
```

**\components\sdCard\CMakeLists.txt**
```
idf_component_register(SRCS "sdCard.cpp" "sdCard.hpp"
                    INCLUDE_DIRS "."
                    REQUIRES ArduinoJson)
```

**\components\sdCard\sdCard.hpp**
```
#ifndef SDCARD
#define SDCARD

#include "ArduinoJson.h"

class sdCard{
    public:
        bool listAudioLib(DynamicJsonDocument &libJson);
};

#endif
```

**\components\sdCard\sdCard.cpp**
```
#include "sdCard.hpp"

bool sdCard::listAudioLib(DynamicJsonDocument &libJson){

    return true;
}
```

### Error Message
```
Linking CXX executable sd_card.elf
FAILED: sd_card.elf 
cmd.exe /C "cd . && D:\Espressif\ESP-IDF-Tools\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\sd_card.elf.rsp -o sd_card.elf  && cd ."
d:/espressif/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x4): undefined reference to `sdCard::listAudioLib(ArduinoJson::V6212PB::BasicJsonDocument<ArduinoJson::V6212PB::DefaultAllocator>&)'
d:/espressif/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj): in function `ArduinoJson::V6212PB::detail::VariantData::VariantData()':
D:/Espressif/esp-idf/components/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:21: undefined reference to `sdCard::listAudioLib(ArduinoJson::V6212PB::BasicJsonDocument<ArduinoJson::V6212PB::DefaultAllocator>&)'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### Environment

- Microcontroller: ESP32-LyraT V4.3
- OS: Windows 10 Pro
- IDE: VS Code with Espressif IDF extension v1.6.3
- Librarys:
         - esp-idf              v4.4.4
         - arduino-esp32  v4.4
         - esp-adf             v2.4
         - ArduinoJson     6.21.2

I don't realy know, where the error comes from. I don't think the error message is pointing to the wright problem.

I hope you have an idea, what the problem is and how to fix that.

I am very grateful that you are helping me with this problem, I am at a loss at the moment.
Thank you very much.

Regards

Fabian

MTBfs, 2023-06-07:
Hello Benoit,

I finally found the error.
I've placed the ArduinoJson library-folder under 'esp-idf/components/' directory but it has to riside in 'esp-idf/components/arduino/libraries' directory.

So now is everything compiling and running fine.

Thank you very much for your help again.

Regards
Fabian

closed 2023-06-07 by MTBfs