bblanchon/ArduinoJson #1928: Serialization issues when not running on main thread

danluca, 2023-06-14:
**Describe the bug**  
When running the JSON serialization on a thread other than main, the action leads to corruption and program hanging (the visible observation), or - most likely - buffer overflow.

I'm a beginner when it comes to Arduino and C++ - so it's a definite possibility I'm doing something horribly wrong. 

The attached sketch (change extension from txt) should illustrate the problem. In its current state, it'll lead to hanging in the Serial monitor no further than writing the first field's value. If we swap the thread allocation - comment/uncomment the matching statements in the sketch (both setup and loop functions) - the JSON serialization works flawlessly.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
The program hangs/crashes before reaching the point of writing `doc.overflowed()` to Serial port - hence cannot complete the troubleshooter.

**Environment**  
Here is the environment that I used:
* Microcontroller: Nano RP 2040
* Core/runtime: Mbed OS Nano 4.0.2 for Arduino Nano boards
* IDE: Arduino IDE 2.1.0

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
#include <ArduinoJson.h>
#include <Scheduler.h>
#include <LibPrintf.h>

uint curFxIndex = 51;
void setup() {
  Scheduler.startLoop(sendJsonSample);
}
[json-thread.txt](https://github.com/bblanchon/ArduinoJson/files/11751202/json-thread.txt)

void loop() {
  mainSample();
}

void sendJsonSample() {
  StaticJsonDocument<512> doc;
  doc["curEffect"] = String(curFxIndex);
  JsonArray fxArray = doc.createNestedArray("fx");
  uint len = serializeJson(doc, Serial);
  printf("Written %d bytes worth of JSON", len);
  delay(5000);
  yield()
}
```

**Compiler output**    
No errors during compilation
```
Sketch uses 98372 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 43572 bytes (16%) of dynamic memory, leaving 226764 bytes for local variables. Maximum is 270336 bytes.
```

**Program output**  
Serial monitor output:
```
17:33:00.076 -> In main thread... idling...
17:33:00.076 -> =========================================
17:33:00.076 -> {"curEffect":"
(nothing more after this)
```
Expected output:

```
18:02:34.129 -> In main thread... idling...
18:02:34.160 -> =========================================
18:02:34.160 -> {"curEffect":"51","fx":[{"name":"fx01","description":"This is a description string for the effect FX01","constantSpeed":true,"dotSize":8,"hiBright":160,"loBright":40,"colorNbr":16,"stackColor":5}]}Written 15663 bytes worth of JSON
18:02:34.206 -> 0
```

bblanchon, 2023-06-16:
Hi @danluca,

This could be a stack overflow.
Can you try with a [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/)?

Best regards,
Benoit

danluca, 2023-06-19:
Hi @bblanchon  - thanks for the suggestion. The code does work with the DynamicJsonDocument object.

I believe you are right about the stack overflow - I have launched the other thread with a larger stack size and the StaticJsonDocument works too: `Scheduler.startLoop(sendJsonSample, 2048)`. On RP2040, the default thread stack size is 1024 bytes. I doubled it and made it work - thank you for the hint :)

I'll close this issue as it is not an issue with the library, the problem lies rather between my keyboard and the monitor 😉

closed 2023-06-19 by danluca

bblanchon, 2023-06-20:
Thank you for the update, @danluca; I'm sure it will help other developers.