bblanchon/ArduinoJson #1929: DeserializationError error

aimethierry, 2023-06-15:
** DeserializationError error**  
Hello, i was trying Deserialization in ArduinoJSON. However, i found some issues, i am passing a non JSON "**{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}{}**" string but deserialzationError is treaating it as JSON. 

**Environment**  
Here is the environment that I used:
* Microcontroller: [e.g. Atmega2560]
* IDE: [e.g. platformIO IDE]

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
#include <ArduinoJson.h>

void setup() {
  // Initialize serial port
  Serial.begin(9600);
  while (!Serial) continue;

  StaticJsonDocument<200> doc;

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}{}";

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, json);

  // Test if parsing succeeds.
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }

  const char* sensor = doc["sensor"];
  long time = doc["time"];
  double latitude = doc["data"][0];
  double longitude = doc["data"][1];

  // Print values.
  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {
  // not used in this example
}
```

bblanchon, 2023-06-16:
Hi @aimethierry,

`deserializeJson()` stops reading at the closing brace `}`.
It's a feature, not a bug, and the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/ok/invalid/brace) could have told you that.

Best regards,
Benoit

ChaoticNeutralCzech, 2023-06-19:
Speaking of `desearializeJson`, I am using VSCodium and PlatformIO, and I included ArduinoJson-v6.21.2.hpp from [Release 6.21.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.21.2). (The same happens if I include ArduinoJson.hpp from  the release's ZIP file, or use PlatformIO's Library Manager to install it as a dependency and then add `#include <ArduinoJson.h>` to the top of `main.cpp`.)  
I am trying to desearialize a downloaded JSON document, which is covered in [your book's excerpt](https://arduinojson.org/book/deserialization_tutorial6.pdf#page=12) and online tutorials, all of which refer to effectively the following code:

    #include "../lib/ArduinoJson-v6.21.2.hpp"
    //skipped many lines, now we're deep in setup(){ after downloading the data
    DynamicJsonDocument doc(16384); //16 kB is upper bound for the document's expected size
    DeserializationError err = deserializeJson(doc, receivedData); //receivedData is a char array acquired using WifiClientSecure

However, I get errors  

    identifier "DynamicJsonDocument" is undefined //aka not declared in this scope
    identifier "DeserializationError" is undefined
    identifier "deserializeJson" is undefined

Did the codebase change *so much* between the book's publication date (2021-04-10) and the current release that the information is no longer relevant? I know some deprecation occured between versions 5 and 6 but the book is specifically called *Mastering ArduinoJson* ***6*** so I thought it would be applicable in version 6.21.2 as well. Which version of ArduinoJson should I include in my projects (and how) to prevent the aforementioned `undefined` errors?

ChaoticNeutralCzech, 2023-06-19:
Turns out, my IDE was being weird. Of course `desearializeJson()` was found after several compilation attempts and restarting the software.

bblanchon, 2023-06-20:
@ChaoticNeutralCzech, thank you very much for purchasing my book ❤️ 
Your problem is that you used `ArduinoJson.hpp` instead of `ArduinoJson.h`. The difference between these headers is that `ArduinoJson.hpp` keeps every declaration in the `ArduinoJson` namespace, whereas `ArduinoJson.h` imports them in the global namespace.
Therefore, you could fix your issue by including `ArduinoJson.h` or by adding `using namespace ArduinoJson;` in the global scope.
Next time, however, please don't hijack an unrelated conversation and [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new/choose) instead.

@aimethierry, is your problem solved? If so, please close this issue.

closed 2023-06-29 by bblanchon