bblanchon/ArduinoJson #1932: const char* and strlcpy to extract json values

philippesikora, 2023-06-17:
I use a DynamicJsonDocument to  store  constant c-strings 

ESP32 reboot when I try to extract value with const char*.
It works well with const char* period_end_str but ESP2 crash with const char* pv_estimate_str
If I replace C-style character arrays const char* pv_estimate_str with String class it works.

DynamicJsonDocument doc(16384); 
DeserializationError error = deserializeJson(doc, http.getStream());  

for (JsonObject forecast : doc["forecasts"].as<JsonArray>()) {

           char period_end[30];              // "2023-04-03T14:00:00.0000000Z"
           const char* period_end_str = forecast["period_end"].as<const char*>();
           Serial.println("debug_step_2 EXTRACT period_end");
           strlcpy(period_end, period_end_str, 30);                            // this code to retrieve period_end works
           Serial.print("debug_step_3 STRLCPY period_end ");
           Serial.println(period_end_str);
           
        
           char pv_estimate[30];           // "1.157"
           const char* pv_estimate_str = forecast["pv_estimate"].as<const char*>();
           Serial.println("debug_step_4 EXTRACT pv_estimate");
           delay(5000);
           Serial.println("still working");
           delay(5000);
           Serial.print("print char array pv_estimate_str");
           Serial.println(pv_estimate_str);          // Guru Meditation Error Core 1 panic'ed LoadProhibited. Exception was unhandled
           strlcpy(pv_estimate, pv_estimate_str, 30);
           Serial.println("debug_step_5 STRLCPY pv_estimate ");
           double dEstimate=atoi(pv_estimate);
           
           /*
           String pv_estimate = forecast["pv_estimate"].as<String>();  // this code to retieve pv_estimate works
           float dEstimate = pv_estimate.toDouble();  
           Serial.println("debug_step_4 EXTRACT with String class pv_estimate");
           */
 }

philippesikora, 2023-06-18:
ESP32 reboot at:  const char* pv_estimate = forecast["pv_estimate"].as<const char*>();

    no issue with :
    String pv_estimate = forecast["pv_estimate"].as<String>();  // this code to retieve pv_estimate works
    float dEstimate = pv_estimate.toDouble();     
     
    *******************************************************************************
       
       for (JsonObject forecast : doc["forecasts"].as<JsonArray>()) {

   
           const char* period_end = forecast["period_end"].as<const char*>();
           
           const char* pv_estimate = forecast["pv_estimate"].as<const char*>();
           double dEstimate=atof(pv_estimate);     // Convert the pv_estimate string to a decimal 
           
           // String pv_estimate = forecast["pv_estimate"].as<String>();  // this code to retieve pv_estimate works
           // float dEstimate = pv_estimate.toDouble();     
            
           char hourStr[3];
           strlcpy(hourStr, period_end + 11, sizeof(hourStr));
           int nHour = atoi(hourStr);

                
           char minuteStr[3];                      // Extracting minutes 
           strncpy(minuteStr, period_end + 14, 2); // Copy the characters at index 14 and 15 (minutes)
           minuteStr[2] = '\0';                    // Null-terminate the string
           int nMinute = atoi(minuteStr);          // Convert the minutes string to an integer

                 
           int nTMins = nHour*60+nMinute;  
           if(nTMins < 30) bTomorrow = true;       // Crossed midnight, get thext 24h of data
      
           if(bTomorrow && (nCount < 48)) {    
                                                   // There are 48 readings in 24h - 
                                                   // so once the first midnight is detected, get the next 48 readings 
             nCount++; 
             dTomorrowTotal = dTomorrowTotal + dEstimate / 2;   
           } 
       }

bblanchon, 2023-06-18:
Hi @philippesikora,

Please share the troubleshooter's report.
https://arduinojson.org/v6/troubleshooter/

Best regards,
Benoit

philippesikora, 2023-06-18:
1. The issue happens at run time
2. The issue concerns deserialization
3. `deserializeJson()` returns `Ok`
4. The strings in the `JsonDocument` change for no reason
5. Input type is `const char*`

philippesikora, 2023-06-18:
The JSON format from Solcast API  is as follow:
{"pv_estimate":0.9028,"pv_estimate10":0.902,"pv_estimate90":0.9028,"period_end":"2023-06-18T15:00:00.0000000Z","period":"PT30M"}
pv_estimate data is not String format, how can I retrieve data with ArduinoJson library?
Thank you
Philippe
https://docs.solcast.com.au/#25ff8ad7-e2a8-44be-9d2e-62e0f73cefd6

philippesikora, 2023-06-18:
Solved with ArduinoJson Assistant
 DynamicJsonDocument doc(24576);  // size: 24576 provided by https://arduinojson.org/v6/assistant/#/step1
const char* period_end = forecast["period_end"].as<const char*>();
double dEstimate = forecast["pv_estimate"].as<double>(); 
You can close this issue
Last question, do you see any problem with heap fragmentation for this code ?
Thank you

******************************************************

void solcast(void) {

  unsigned long startTime;
  unsigned long endTime;
  unsigned long routineTime;
  double dTomorrowTotal_kWh = 0.0;
 
  startTime = millis(); 
  
  Serial.println("task_1 Solcast");
   
  if (WiFi.status() == WL_CONNECTED) {

     HTTPClient http;
        
     const size_t authHeaderSize = 50;
     char authHeader[authHeaderSize];

     http.useHTTP10(true);   // Use HTTP/1.0 protocol
     http.begin(url);        // Make HTTP request

     // Add User-Agent header
     http.addHeader("User-Agent", "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36 Firefox/87.0");
     http.addHeader("Accept-Encoding", "gzip, deflate, br");
     snprintf(authHeader, authHeaderSize, "Bearer %s", apiKey);
     http.addHeader("Authorization", authHeader);
     
     
     int httpCode = http.GET();              /* start connection and send HTTP header */ 
     xQueueOverwrite(xMailbox_4, &httpCode); // send httpCode response code to xMailbox_4
     
     Serial.print("HTTP response code: ");  // if response code = 429 Too Many Requests 
     Serial.println(httpCode);              //                    401 lacks valid authentication credentials for the target resource
  
     if (httpCode > 0) {   

       
       
       DynamicJsonDocument doc(24576);  // size: 24576 provided by https://arduinojson.org/v6/assistant/#/step1
       //StaticJsonDocument<24576> doc; // dos not work in static
          
       DeserializationError error = deserializeJson(doc, http.getStream());  
        
       if (error) {                                      // Test if parsing succeeds
         Serial.print("deserializeJson() failed: ");
         Serial.println(error.c_str());
         return;
       }

       bool bTomorrow = false;
       int nCount = 0;
       
       Serial.println("debug_step_1 deserializeJson");      
       
       for (JsonObject forecast : doc["forecasts"].as<JsonArray>()) {    
       
           const char* period_end = forecast["period_end"].as<const char*>();
           double dEstimate = forecast["pv_estimate"].as<double>(); 
    
           char hourStr[3];                                       // Destination array
           strlcpy(hourStr, period_end + 11, sizeof(hourStr));    // Copy the characters at index 11
           int nHour = atoi(hourStr);                             // Convert the minutes string to an integer
 
           char minuteStr[3];                                     // Destination array
           strlcpy(hourStr, period_end + 14, sizeof(minuteStr));  // Copy the characters at index 14
           int nMinute = atoi(minuteStr);                         // Convert the minutes string to an integer

                       
           int nTMins = nHour*60+nMinute;  
           if(nTMins < 30) bTomorrow = true;       // Crossed midnight, get thext 24h of data
      
           if(bTomorrow && (nCount < 48)) {    
                                                   // There are 48 readings in 24h - 
                                                   // so once the first midnight is detected, get the next 48 readings 
             nCount++; 
             dTomorrowTotal_kWh = dTomorrowTotal_kWh + dEstimate / 2;   
           } 
       }
             
       
     } else {
       Serial.println("HTTP request failed");
       } 

     http.end();
     endTime = millis(); 
     routineTime = (endTime - startTime); // according to test routineTime = 2.3s
     
     xQueueOverwrite(xMailbox_1, &dTomorrowTotal_kWh);  // send dTomorrowTotal to xMailbox_1
     xQueueOverwrite(xMailbox_2, &routineTime);         // send routineTime to xMailbox_2
   
     Serial.println(); 
     Serial.print("solcast_tomorrow "); Serial.print(dTomorrowTotal_kWh); Serial.print(" kWh"); Serial.println(); 
     Serial.print("Routine_time "); Serial.print(routineTime); Serial.print(" ms"); Serial.println();     
  }   
}

bblanchon, 2023-06-19:
Hi @philippesikora,

I'm glad you found a solution.

I don't see anything that should fragment the heap in your code snippet.
[Heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) occurs when you allocate/deallocate blocks of variable size (typically `String`s), which is not the case here.

BTW, you can leverage the fact that [`atoi()`](https://en.cppreference.com/w/cpp/string/byte/atoi) stops reading at the first non-numeric character, to simplify the date-parsing part:

```c++
int nHour = atoi(period_end + 11);
int nMinute = atoi(period_end + 14);
```

Best regards,
Benoit

closed 2023-06-19 by bblanchon