bblanchon/ArduinoJson #1934: Deserialized double has only 6 digit precision

margaretselzer, 2023-06-20:
Using  ArduinoJson @ 6.19.4

I want to deserialize a double value, but the precision is only 6 digits when I assign it to a double variable. The same goes for float where I would expect 7 digits. I think I read somewhere that ArduinoJson supports up to 9 digits precision, so I would expect to be able to get that out of it. Do I do something wrong?

double myDouble = jsonDoc["mydouble"];  ==> only six digits precision
float myFloat = jsonDoc["mydouble"];  ==> only six digits precision
std::cout << jsonDoc["mydouble"]; ==> 9 digits precision

It appears that if I deserialize to std::ostream only then I get the full precision, but I would not want to use std::ostream as a proxy just to get the desired output (not even sure right now how I would do that).

Here is a small snippet that demonstrate the problem: 
See also: https://wandbox.org/permlink/jdXYFiDipCAjS0ic

```c++
#include <iostream>
#define ARDUINOJSON_USE_DOUBLE 1
#include "ArduinoJson.h"

int main() {
  const char jsonDataForGainTest[107] = "{ \"id\" : 65535, \"dsaId\" : 65535, \"rfCentFreq\" : 4294967295, \"nominalAtt\" : 65535, \"gain\" : 3.14159265359 }"    ;
  StaticJsonDocument<1024> tmpDoc;
  deserializeJson(tmpDoc, jsonDataForGainTest);
    
  double myDoubleGain = tmpDoc["gain"].as<double>();
  float myFloatGain = tmpDoc["gain"];
    
  std::cout << tmpDoc["gain"] << std::endl;
  std::cout << myDoubleGain << std::endl;
  printf("GAIN TEST myDoubleGain: %f, myFloatGain: %f, tmpDoc[\"gain\"]: %f", myDoubleGain, myFloatGain, static_cast<double>(tmpDoc["gain"]));
   
  return 0;
}
```

**Program output**  
Expected output:
```
3.141592654 
3.141592654
GAIN TEST myDoubleGain: 3.141592654, myFloatGain: 3.1415927, tmpDoc["gain"]: 3.141592654
```

Actual output:
```
3.141592654
3.14159
GAIN TEST myDoubleGain: 3.141593, myFloatGain: 3.141593, tmpDoc["gain"]: 3.141593
```

margaretselzer, 2023-06-20:
Update: 
If I use instead of 3.14159265359 333.14159265359, the difference in behavior becomes even more obvious. The result of the same code will be: 
```
333.1415927
333.142
GAIN TEST myDoubleGain: 333.141593, myFloatGain: 333.141602, tmpDoc["gain"]: 333.141593
```
It appears so that in case of float, the total number of supported digits (including the decimal dot) is 7 while in case of double it is 9 excluding the decimal dot. This would really limit the usability of ArduinoJson in case of floating point numbers.

margaretselzer, 2023-06-20:
Well, I think I now understand what happens. See the code below.

First of all std::cout default precision is 6. That explains the result when double and float variables are printed.
Second - although I did not find the code - it appears that ArduinoJson implements operator<<() for ostream so that it converts the JSON value to string, so precision has nothing to do with the output.
Third, jsonDoc["field"] by default returns a float for any floating point even if the content is a double. This I did not expect, but good to know.

Conclusion: one should always use `.as<T>()` instead of being content with just operator[] on the json doc. 

```
#include <iostream>
#include <iomanip>
#include <limits>
#include <string>
#define ARDUINOJSON_USE_DOUBLE 1
#include "ArduinoJson.h"

int main() {
  const char jsonDataForGainTest[109] = "{ \"id\" : 65535, \"dsaId\" : 65535, \"rfCentFreq\" : 4294967295, \"nominalAtt\" : 65535, \"gain\" : 333.14159265359 }"    ;
  StaticJsonDocument<1024> tmpDoc;
  deserializeJson(tmpDoc, jsonDataForGainTest);
    
  double myDoubleGain = tmpDoc["gain"].as<double>();
  float myFloatGain = tmpDoc["gain"];
    
  std::cout << tmpDoc["gain"] << " | " << tmpDoc["gain"].as<std::string>() << " | " << tmpDoc["gain"].as<double>() << " | " << std::setprecision(std::numeric_limits<float>::digits10) << tmpDoc["gain"].as<float>() << " | " << std::setprecision(std::numeric_limits<double>::digits10) << tmpDoc["gain"].as<double>() << std::endl;
  std::cout << myDoubleGain << " | " << std::setprecision(std::numeric_limits<double>::digits10) << myDoubleGain << std::endl;
  printf("GAIN TEST myDoubleGain: %.15f, myFloatGain: %.7f, tmpDoc[\"gain\"]: %.15f", myDoubleGain, myFloatGain, static_cast<double>(tmpDoc["gain"]));
   
  return 0;
}
```
Produces the following result:
```
333.1415927 | 333.1415927 | 333.142 | 333.142 | 333.14159265359
333.14159265359 | 333.14159265359
GAIN TEST myDoubleGain: 333.141592653590067, myFloatGain: 333.1416016, tmpDoc["gain"]: 333.141592653590067
```

closed 2023-06-20 by margaretselzer

bblanchon, 2023-06-21:
Hi @margaretselzer,

This issue is not due to ArduinoJson but to `printf()`:

```c++
printf("%f", 3.14159265359);    // 3.141593
printf("%.9f", 3.14159265359);  // 3.141592654
```

Online demo: https://wandbox.org/permlink/GQ5mNE5cAYStIUSP

Best regards,
Benoit