bblanchon/ArduinoJson #1937: How to convert JsonDocument to a string or a char array?

KiraVerSace, 2023-06-24:
OK, this is my sourc code
```
StaticJsonDocument<512> docTemp;

docTemp.clear();
deserializeJson(docTemp, xSensorData.externChannel[0].data);

params_sensor_value["ch1"] = docTemp;
```

Then you can see the reuslt json, it's all right
```
{
    "sensor": {
        "value": {
            "ch1": {
                "status": "success",
                "waterLevel": "0.508"
            }
        }
    }
}
```

but my cloud server do not support JSON nesting , so I should upload my json like this:
```
{
    "sensor": {
        "value": {
            "ch1": ""status": "success","waterLevel": "0.508""
        }
    }
}
```

Is there any way easily to exchange it?

Thank you for your help!

[image]

bblanchon, 2023-06-27:
Hi @KiraVerSace,

You know the document you want to generate is not a valid JSON document, right?
You should fix your server, but if you cannot, you may be able to generate this ill-formatted document with [serialized()](https://arduinojson.org/v6/api/misc/serialized/).

Best regards,
Benoit

KiraVerSace, 2023-06-28:
Thank you for your help.

closed 2023-06-28 by KiraVerSace